RobotLocomotion/drake #20176: Compliant meshes (as vtk file) are missing from the

jwnimmer-tri, 2023-09-12:
(This ticket was broken out of #18913 as a separate item.

**Is your feature request related to a problem? Please describe.**

In the [Hydroelastic User Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html), there's no mention of introducing compliant meshes (as vtk file)

**Describe the solution you'd like**

This is obviously something that would be worthwhile to note in the user guide, or maybe to mention in the user guide with a citation to an interactive tutorial on VTK file creation.

**Describe alternatives you've considered**

The `hydroelastic_contact_basics.ipynb` tutorial has one brief mention:

> For non-convex objects, each mesh must be provided as a VTK file (not a .obj file) containing a tetrahedral mesh to use them as compliant-hydroelastic geometries

That does not really explain it very well.

DamrongGuoy, 2023-09-14:
Remind myself that in addition to [Hydroelastic User Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html), we'll have geometry/geometry_file_formats_doxygen.h (#20173).

jwnimmer-tri, 2024-03-20:
I know we recently updated our tutorials in this area.  What _specifically_ more do we want to document and _where_?

User guide?  Geometry file formats?  etc.

DamrongGuoy, 2024-03-20:
We can close it now.

closed 2024-03-20 by DamrongGuoy