RobotLocomotion/drake #20180: [Meshcat] Add option to set camera position only (instead of entire Pose)

Vinggui, 2023-09-13:
The new Meshcat API allows us to set/move the default camera pose, or just its target point. I imagine these two functions will cover most of the user's needs, but eventually setting just the camera position could also be useful.

**Describe the solution you'd like**
I think that Drake could offer one more API option to set just the camera position (e.g., `SetCameraPosition()`). I know it can be currently done with `SetProperty()`, but its usage it not intuitive and it required Z-up rotation.

Having one more API call consistent with the design of `SetCameraPose()` could be more reliable. A possible modification would be:
```
  void SetCameraTarget(const Eigen::Vector3d& p_WT,
                       bool only_perspective = true) {
....
  }

  // This function is public via the PIMPL.
  void SetCameraPosition(const Eigen::Vector3d& p_WC) {
    SetTransform("/Cameras/default", math::RigidTransformd());
    // The camera position in meshcat's y-up world.
    const Eigen::Vector3d p_WC_y = MeshcatYUpPosition(p_WC);
    SetProperty("/Cameras/default/rotated/<object>", "position",
                std::vector<double>{p_WC_y.x(), p_WC_y.y(), p_WC_y.z()});
  }

  // This function is public via the PIMPL.
  void SetCameraPose(const Eigen::Vector3d& p_WC, const Eigen::Vector3d& p_WT) {
    SetCameraTarget(p_WT, false /* only_perspective */);
    SetCameraPosition(p_WC);
  }
```

**Describe alternatives you've considered**
As [mentioned](https://github.com/RobotLocomotion/drake/pull/19968#issuecomment-1690740240) by @SeanCurtis-TRI , it could also be accomplished by making the the target parameter (p_WT) optional in [SetCameraPose()](https://github.com/RobotLocomotion/drake/blob/c482ed7f6222958eb1f2904db9bbbd22e6eedf9f/geometry/meshcat.cc#L1310).

Thank you!

SeanCurtis-TRI, 2023-09-13:
To add a further elaboration: the semantics of setting camera *position* without setting the target position would move the camera, but rotate it to keep it looking at the original target.

Xining-Du, 2023-09-13:
@SeanCurtis-TRI I assume we will need to add the `GetCameraTarget` API then?

Vinggui, 2023-09-13:
> @SeanCurtis-TRI I assume we will need to add the `GetCameraTarget` API then?

Or the entire Pose?

SeanCurtis-TRI, 2023-09-13:
@Xining-Du can you elaborate on why you would request camera target? 

During the original camera posing PR, we also speculated about an API where you set the camera pose via `RigidTransform` (setting its position *and* orientation) and could provide an optional target *distance* (such that your camera is positioned and oriented as you've specified and Drake handles setting the target point to be consistent with that.) I suspect, this kind of behavior would solve the same problem and, possibly, be preferred. But I'd like to know what problem you have in mind before I can say that with certainty.

Xining-Du, 2023-09-13:
@SeanCurtis-TRI Ah, I might have misunderstood how camera pose works. Was thinking that Drake would need to get the current camera target, and use the input camera position to figure out the orientation. However from your comments, it seems that this is all handled by meshcat automatically, i.e. If we only set the camera position using `SetProperty` meshcat API, then the camera will automatically point to the current target in meshcat?

Vinggui, 2023-09-13:
> @SeanCurtis-TRI Ah, I might have misunderstood how camera pose works. Was thinking that Drake would need to get the current camera target, and use the input camera position to figure out the [...]

Currently, I believe that setting the position separately (e.g, using `SetProperty()`) will indeed keep the camera looking (targeted) to the same spot, which is why @SeanCurtis-TRI added this [comment](https://github.com/RobotLocomotion/drake/issues/20180#issuecomment-1717913600). For me, this behavior is perfect.

However, currently as well, `SetCameraPose()` needs a Target, so I can not ignore (nor acquire the current camera values - position, target, zoom). Therefore, for what we need **now**, a `SetCameraPosition()` (or similar) would be enough.

A variant of `SetCameraPose()` with `RigidTransform` would maybe be convenient, but I believe it would not help us do the position change without changing the target (at least not in a simple way).

Now, adding a `GetCameraPose()` or `GetCameraParameters()` would allow us to easily and programmatically store the current scene (it could also go into the `staticHTML`).

SeanCurtis-TRI, 2023-09-13:
So, here's my summary:

1. To move the camera and maintain the view target
    - The simplest way is to modify `SetCameraPose()` so that the `target_in_world` parameter is optional.
2. To position and orient the camera arbitrarily:
    - I think the best way is to the `SetCameraPose(RigidTransform)` solution.
    - We'd redefine the view target based on a `target_distance` which is defaulted to 1 meter, but can be specified.
    - This would get your camera  posed the way you want, and keep the interactive camera control behaving itself.
        - If you never needed to interact with it, you wouldn't notice if the orbit controller was broken or not.

Further thoughts:

In the near term, the intention is for you to be able to configure the meshcat camera via `MeshcatParams` so you can specify it in a scenario.yaml file or some such thing.