RobotLocomotion/drake #20222: Feasibility of simulation of an unmanned surface vehicle (USV) with a manipulator

IamGyal, 2023-09-20:
Hi,

I'm working on a custom unmanned surface vehicle (USV) with a waterproof manipulator attached to it.

I'm looking for a simulator where I can validate the maths models of the robot. 

The ideal case would be where
- I do not need to put equations to calculate the hydrodynamics in the simulator,
- but I can just put the geometry of my robot, force and torque inputs, and any other necessary parameters
- and see that both the USV and the attached manipulator are affected by corresponding hydrodynamics (buoyancy, drag, lift, and added mass)
- and the forces and torques (wrenches) propagated between the manipulator and the USV are also taken into account
- and the combined system (the USV and the manipulator) moves/behaves accordingly and accurately.

Could this be realised with Drake? Or any other suggestions and advice would be appreciated.

Thank you very much.

xuchenhan-tri, 2023-09-20:
Currently Drake simulation doesn't model hydrodynamics. However, you should be able to model the physics to the fidelity of your requirement yourself with the system framework infrastructure provided by Drake. If you can leverage other components (planning, control, etc) in Drake at the same time, then using Drake can definitely get you a leg up.

\cc @sherm1 @amcastro-tri in case you have thoughts to share on this.

Assigning @sherm1 for triage/reassignment per platform reviewer checklist.

closed 2023-09-20 by sherm1

RussTedrake, 2023-09-21:
See also https://github.com/RobotLocomotion/drake/issues/19440.