RobotLocomotion/drake #20226: Mesh/convex semantics for vtk files are maybe incomplete

rpoyner-tri, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
We have minimal support for .vtk files as a format for Mesh and Convex geometry. It is not clear that the current semantics are as complete as they could be.

* It is not clear that we can do anything sensible with a vtk file that is *not* a tet mesh.
* It is not clear that we reliably report errors when we get a vtk file with data we don't support.
* The code for building hydroelastic representations doesn't support the full matrix of geometry and hydroelastic type declarations.

Assuming we have a vtk containing a valid tet mesh, this table estimates the current results (some test cases pending):

| Hydro Type | Mesh | Convex |
|------------|------|--------|
| compliant | volume | volume |
| rigid | surface | Crash? (passed to ReadObjFile) | 
| unknown | throw | throw |

**Describe the solution you'd like**
Instead:
* we should at least give a reasonable error message for non-test-mesh vtk files (there are many possible types).
* we should maybe do something better in the matrix entries labeled Crash or throw above.
* we should document the supported behaviors.

**Describe alternatives you've considered**
We could do nothing, and suppose the current situation is good enough.

**Additional context**
Reducing the number of paths that crash or throw will potentially make future support for #19709 smoother and more compelling.

rpoyner-tri, 2023-09-20:
Assigning @DamrongGuoy (as component owner) for visibility; I'll likely work on this shortly.

rpoyner-tri, 2023-09-20:
\cc @SeanCurtis-TRI for closure of the prior conversation.

DamrongGuoy, 2023-09-21:
In case this is relevant, at the low level, these are related components:
//geometry/proximity:make_mesh_from_vtk
//geometry/proximity:vtk_to_volume_mesh
//geometry/proximity:make_mesh_field

If the higher-level code doesn't catch the problem, the reading of VTK file will happen in vtk_to_volume_mesh, where we can try to give reasonable error messages for different cases of non-tet-mesh vtk files.

I have seen users use a VTK file containing a triangle mesh instead of a tetrahedral mesh for compliant-hydroelastic and got this error message:
```
2023-09-12 18:30:03.736 (   0.333s) [        2C7BA000]vtkUnstructuredGridRead:99     ERR| vtkUnstructuredGridReader (0x5644726655e0): Cannot read dataset type: polydata
```

rpoyner-tri, 2023-10-31:
Claiming this is resolved after the two PRs noted above.

closed 2023-10-31 by rpoyner-tri