RobotLocomotion/drake #20251: [ci] Prefer Jammy for most builds

jwnimmer-tri, 2023-09-25:
Most all of our developers use Jammy now, and we'll be dropping Focal support in ~6 months.  As such, when we're only running one flavor of some build (e.g., Address Sanitizer") we should run on Jammy (only), not Focal (only).  The lists are a little bit difficult to reconcile with continuous vs nightly etc, but here's my best take:

---

Switch this pre-merge build (and its Continuous twin) to Jammy for better a sanity checking cover:
- linux-focal-clang-bazel-experimental-everything-release
- linux-focal-clang-bazel-experimental-continuous-release

All coverage should switch to Jammy (currently is all Focal):
- linux-focal-clang-bazel-nightly-coverage
- linux-focal-clang-bazel-weekly-everything-coverage
- linux-focal-clang-bazel-weekly-gurobi-coverage
- linux-focal-clang-bazel-weekly-mosek-coverage
- linux-focal-clang-bazel-weekly-snopt-coverage
- linux-focal-gcc-bazel-nightly-coverage
- linux-focal-gcc-bazel-weekly-everything-coverage
- linux-focal-gcc-bazel-weekly-gurobi-coverage
- linux-focal-gcc-bazel-weekly-mosek-coverage
- linux-focal-gcc-bazel-weekly-snopt-coverage

Focal-only sanitizers / memcheck should switch to Jammy-only:
- linux-focal-clang-bazel-continuous-everything-address-sanitizer
- linux-focal-clang-bazel-continuous-leak-sanitizer
- linux-focal-clang-bazel-nightly-address-sanitizer
- linux-focal-clang-bazel-nightly-everything-leak-sanitizer
- linux-focal-clang-bazel-nightly-everything-thread-sanitizer
- linux-focal-clang-bazel-nightly-everything-undefined-behavior-sanitizer
- linux-focal-clang-bazel-nightly-leak-sanitizer
- linux-focal-clang-bazel-nightly-thread-sanitizer
- linux-focal-clang-bazel-nightly-undefined-behavior-sanitizer
- linux-focal-clang-bazel-nightly-everything-valgrind-memcheck
- linux-focal-clang-bazel-nightly-valgrind-memcheck

Weekly spot-checks should switch from Focal-only to Jammy-only:
- linux-focal-clang-bazel-weekly-gurobi-debug
- linux-focal-clang-bazel-weekly-mosek-debug
- linux-focal-clang-bazel-weekly-snopt-debug
- linux-focal-gcc-bazel-weekly-gurobi-debug
- linux-focal-gcc-bazel-weekly-gurobi-release
- linux-focal-gcc-bazel-weekly-mosek-debug
- linux-focal-gcc-bazel-weekly-mosek-release
- linux-focal-gcc-bazel-weekly-snopt-debug
- linux-focal-gcc-bazel-weekly-snopt-release
- linux-focal-gcc-cmake-weekly-gurobi-debug
- linux-focal-gcc-cmake-weekly-gurobi-release
- linux-focal-gcc-cmake-weekly-mosek-debug
- linux-focal-gcc-cmake-weekly-mosek-release
- linux-focal-gcc-cmake-weekly-snopt-debug
- linux-focal-gcc-cmake-weekly-snopt-release

Dual Focal+Jammy sanitizers / memcheck should switch to Jammy-only:
- Listing TBD.  Mostly I think it's just the "weekly" stuff.

The continuous unprovisioned might as well go up to Jammy:
- linux-focal-unprovisioned-gcc-bazel-continuous-release

The CI tools should probably upgrade now to Jammy?
- linux-focal-unprovisioned-gcc-bazel-continuous-mirror-to-s3
- linux-focal-nightly-cache-server-health-check
- linux-focal-continuous-cache-server-health-check

---

The continuous packaging canary can _remain on Focal_ (our oldest platform):
- linux-focal-unprovisioned-gcc-bazel-continuous-packaging

BetsyMcPhail, 2023-10-11:
Still left to convert: `linux-focal-clang-bazel-experimental-everything-release` and `linux-focal-clang-bazel-experimental-continuous-release`. I don't anticipate any issues with those, but I also don't want to convert and leave the office for the day. I will plan on doing the switch tomorrow.

Also outstanding is  the question of which dual Focal+Jammy sanitizers / memcheck should switch to Jammy-only.

jwnimmer-tri, 2023-10-11:
> > Dual Focal+Jammy sanitizers / memcheck should switch to Jammy-only:
> > Listing TBD. Mostly I think it's just the "weekly" stuff.
>
> Also outstanding is the question of which dual Focal+Jammy sanitizers / memcheck should switch to Jammy-only.

Yeah, sorry, that was unclear.  The goal is is that for Sanitizers and Memcheck that all Production jobs are Jammy-only.  So in cases like `linux-{focal,jammy}-clang-bazel-weekly-snopt-valgrind-memcheck` and `linux-{focal,jammy}-clang-bazel-weekly-gurobi-address-sanitizer` and etc we should just remove the Weekly Focal job.  So the TBD wasn't about if we should keep any on Focal, but rather I just hadn't copy-pasted the full list yet.

BetsyMcPhail, 2023-10-12:
Thanks for the clarification, I can make the change without the full list.

BetsyMcPhail, 2023-10-12:
The following weekly Focal jobs will be removed (the equivalent Jammy job already exists):

- gurobi-address-sanitizer
- gurobi-leak-sanitizer
- gurobi-thread-sanitizer
- gurobi-undefined-behavior-sanitizer
- gurobi-valgrind-memcheck
- mosek-address-sanitizer
- mosek-leak-sanitizer
- mosek-thread-sanitizer
- mosek-undefined-behavior-sanitizer
- mosek-valgrind-memcheck
- snopt-address-sanitizer
- snopt-leak-sanitizer
- snopt-thread-sanitizer
- snopt-undefined-behavior-sanitizer
- snopt-valgrind-memcheck

closed 2023-10-12 by BetsyMcPhail