RobotLocomotion/drake #20256: [framework] Publish events at (tₑ, x⁺(tₑ)) instead of (tₑ, x⁻(tₑ))

jwnimmer-tri, 2023-09-26:
**Is your feature request related to a problem? Please describe.**

Refer to the [Simulator overview](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_simulator.html) documentation for background reading.

Currently, publishes events always use (tₑ, x⁻(tₑ)) as their Context. I have a case where things get a lot simpler for me if I can publish at (tₑ, x⁺(tₑ)) instead. In short, I have a system that declares a concurrent timed unrestricted update event and timed publish event, and I'd like to publish the post-update state at tₑ, instead of at some unknown future point after the world has evolved tₑ+ω.  (More specifically, I'd like a ZOH update event to capture a RgbdSensor output camera image, and a simultaneous publish event to publish that image during the same step.)

Note of course that selecting x⁺ vs x⁻ only applies to publish event declarations -- update events wouldn't/shouldn't/couldn't offer this.

Anyone coming from a background without continuous state (e.g., most RL researchers) will be extremely surprised about our current default (and only) publish event timing.  Anyone making logs or traces from a Simulator today where they record the timestamp + data of their publish events (e.g., with VectorLogSink) will be surprised to find that their logs are all off-by-one in the time versus the published data.

**Describe the solution you'd like**

When declaring a publish event, the user should be able to declare that it publishes using x⁺ not x⁻.  I would love it the that was our new default (I'd argue it's the least surprising choice), but for a while it will need to be opt-in so that we don't break any existing simulations.

**Describe alternatives you've considered**

There is really no other way to accomplish this goal in a robust and composable way.  The x⁺ vs x⁻ is a fundamental modeling choice.

**Additional context**

Discussed in detail [on drakedevelopers slack](https://app.slack.com/client/T0JNB2DS4/C2CHRT98E/thread/C2CHRT98E-1687876610.604139).

jwnimmer-tri, 2023-12-04:
I studied this a bit more.

(1) Tentatively, I think only timed publish events need to distinguish x⁻ vs x⁺.  So for witness-triggered publish, initialize-triggered publish, and force-triggered publish, there are no changes.

(2) For timed publish events, I think my proposal is to add a flag to choose pre-update, post-update, or both.  Something like the following outline:

```patch
diff --git a/systems/framework/event.h b/systems/framework/event.h
index 2a908539d0..621d6cf32b 100644
--- a/systems/framework/event.h
+++ b/systems/framework/event.h
@@ -401,6 +401,12 @@ enum class TriggerType {
  */
 using TriggerTypeSet = std::unordered_set<TriggerType, DefaultHash>;
 
+enum class HybridPublish {
+  kPreUpdate = 1,
+  kPostUpdate = 2,
+  kPreAndPostUpdate = 3,
+};
+
 /** Abstract base class that represents an event. The base event contains two
 main pieces of information: an enum trigger type and an optional attribute
 that can be used to explain why the event is triggered.
diff --git a/systems/framework/leaf_system.h b/systems/framework/leaf_system.h
index c4a27a369f..d63ee429d4 100644
--- a/systems/framework/leaf_system.h
+++ b/systems/framework/leaf_system.h
@@ -281,7 +281,8 @@ class LeafSystem : public System<T> {
   template <class MySystem>
   void DeclarePeriodicPublishEvent(
       double period_sec, double offset_sec,
-      EventStatus (MySystem::*publish)(const Context<T>&) const) {
+      EventStatus (MySystem::*publish)(const Context<T>&,
+      HybridPublish hybrid_publish = HybridPublish::kPreUpdate) const) {
     static_assert(std::is_base_of_v<LeafSystem<T>, MySystem>,
                   "Expected to be invoked from a LeafSystem-derived System.");
     DRAKE_DEMAND(publish != nullptr);
@@ -307,7 +308,8 @@ class LeafSystem : public System<T> {
   template <class MySystem>
   void DeclarePeriodicPublishEvent(double period_sec, double offset_sec,
                                    void (MySystem::*publish)(const Context<T>&)
-                                       const) {
+                                       const,
+      HybridPublish hybrid_publish = HybridPublish::kPreUpdate) {
     static_assert(std::is_base_of_v<LeafSystem<T>, MySystem>,
                   "Expected to be invoked from a LeafSystem-derived System.");
     DRAKE_DEMAND(publish != nullptr);
@@ -587,7 +589,8 @@ class LeafSystem : public System<T> {
   @see Simulator::set_publish_every_time_step() */
   template <class MySystem>
   void DeclarePerStepPublishEvent(
-      EventStatus (MySystem::*publish)(const Context<T>&) const) {
+      EventStatus (MySystem::*publish)(const Context<T>&) const,
+      HybridPublish hybrid_publish = HybridPublish::kPreUpdate) {
     static_assert(std::is_base_of_v<LeafSystem<T>, MySystem>,
                   "Expected to be invoked from a LeafSystem-derived System.");
     DRAKE_DEMAND(publish != nullptr);
```

@sherm1 and I to discuss this week.

sherm1, 2023-12-04:
Maybe "PublishTiming" instead of "HybridPublish" for the enum?

sherm1, 2025-02-05:
Per f2f discussion, a minor correction to the spec above:

The current behavior looks like this: 
  time t -> unrestricted update(t) -> discrete update(t) -> advance time -> publish(t + dt)
We want a "time t" publish option:
  time t -> unrestricted update(t) -> discrete update(t) **-> pre_publish(t)** -> advance time -> post_publish(t + dt)

In the terminology of [Simulator Step documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_simulator.html):
```c++
procedure Step(tₛ, x⁻(tₛ), tₘₐₓ)
  // Update any variables (no restrictions).
  x*(tₛ) ← DoAnyUnrestrictedUpdates(tₛ, x⁻(tₛ))
  // ----------------------------------
  // Time and state are at {tₛ, x*(tₛ)}
  // ----------------------------------
  // Update discrete variables.
  xd⁺(tₛ) ← DoAnyDiscreteUpdates(tₛ, x*(tₛ))
  xc⁺(tₛ) ← xc*(tₛ)  // These values carry over from x*(tₛ).
  xa⁺(tₛ) ← xa*(tₛ)
  // ----------------------------------
  // Time and state are at {tₛ, x⁺(tₛ)}
  // ----------------------------------

  DoAnyPublishes(tₛ, x⁺(tₛ))     // <-- PrePublish goes here; note time & state
```

A question for discussion: should there also be a "pre-monitor" option? The monitor function is just a Diagram-level publish and is currently issued along with the (post) publish.

sherm1, 2025-02-06:
Implementation note:
Looking at the AdvanceTo code, the (maybe non-intuitive) modification we want to make is to modify the behavior here:
```c++
      // At this point merged_events_ has any publish events we need to handle
      // now (including per-step, witnessed, and time-triggered). These are
      // simultaneous and all will be handled even if there is a failure.

      accumulated_event_status.KeepMoreSevere(
          HandlePublish(merged_events_->get_publish_events()));

      if (get_publish_every_time_step()) {
        accumulated_event_status.KeepMoreSevere(
            HandlePublish(system_.get_forced_publish_events()));
      }
```
to _not_ issue the PrePublish events. That way they remain on the timed_events list to be issued at the beginning of the next step. Related code in Initialize() should change in the same way. Then when we add a HandlePublish after Unrestricted & Discrete updates, it will handle publishes that triggered at the end of the previous step.

I think the easiest way to make this change would be
- add a flag to every publish event to indicate whether it is to be handled at beginning or end of step, and
- add a flag to HandlePublish() to tell it whether to issue bos or eos events. Both kinds will be present on the list of events it is given.
- add a HandlePublish(BeginningOfStep) call after discrete events are handled.

sherm1, 2025-02-06:
Terminology suggestion:
PublishTiming: BeginningOfStep, EndOfStep

I think that would be easier for users to understand than pre- and post-publish.

jwnimmer-tri, 2025-04-30:
See a [recent stackoverflow post](https://stackoverflow.com/questions/79600666/how-do-i-make-vectorlogsink-read-the-post-update-output-of-my-discrete-time-cont) for another point -- once we have this feature, we should be sure to amend our `VectorLogSink` to offer an option for which publish timing is desired.