RobotLocomotion/drake #20258: Build error with XCode 15

jwnimmer-tri, 2023-09-26:
### What happened?

The libclang ABI has changed in XCode 15, and Drake's documentation parser doesn't understand it yet.

```
  File ".../mkdoc.runfiles/clang_cindex_python3_internal/clang/cindex.py", line 650, in from_id
    raise ValueError('Unknown template argument kind %d' % id)
ValueError: Unknown template argument kind 350
```

### Version

v1.21.0

### What operating system are you using?

macOS 13 (Ventura)

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2023-09-26:
Drake should probably switch to use https://pypi.org/project/clang/ instead of https://github.com/wjakob/clang-cindex-python3.

jwnimmer-tri, 2023-09-26:
Well, switching would make things more clear but doesn't actually solve the problem.

The libclang enums [broke ABI](https://github.com/llvm/llvm-project/commit/bb83f8e70bd1d56152f02307adacd718cd67e312#diff-bb9efdb13afd008baec9ab1d13bf90f6f78ef314ae65db4be44a94820bb736c4) so the fundamental problem is needing to use a contemporaneous `clang.cindex` to the `libclang.so` being used, but the python is coming from our workspace yet the shared library is coming from the host OS.

jwnimmer-tri, 2023-09-26:
Possible solutions:

(1) Commit multiple copies of `bindings/python/clang` python source code into Drake, and have our `libclang_setup.py` figure out which one to use (and import it).

(2) Download libclang on the fly, either with a full toolchain https://github.com/grailbio/bazel-toolchain or just write our own repository rule for just the binaries we need.  Then, we have full control over which version to use (instead of relying on the host OS, which in the case of Apple software will approximately always be in a broken or about-to-break state).

(3) Switch our Doxygen build to use libclang as its parser, and have Doxygen output the docstrings instead of relying on the fundamentally broken `mkdoc` approach.

svenevs, 2023-09-26:
> (2) Download libclang on the fly, either with a full toolchain https://github.com/grailbio/bazel-toolchain or just write our own repository rule for just the binaries we need.

FWIW the `clang-format` from https://github.com/RobotLocomotion/homebrew-director/blob/master/Formula/clang-format%4012.rb is one of the few left in that repo we still depend on (for macOS only, obviously...).  So if this is the route taken, we should be able to stop maintaining that as well (I think that just leaves gurobi, which won't ever go away).

jwnimmer-tri, 2023-09-26:
For the moment, because we did not have any canary set up to catch this (nor did we manually test the RCs), we are in fire drill mode so I'm going to take a stab at option (1) to put out the fire.  Longer term, I hope to explore the other options.

jwnimmer-tri, 2023-09-27:
As of #20261, the problems seem to be fixed for macOS _users_ who install from source.  There remain problems for Drake Developers who need to build & run Drake's unit tests.

jwnimmer-tri, 2023-10-10:
I'd like to keep this issue pinned until Drake has a stable release with the fix.

jwnimmer-tri, 2023-10-16:
I guess people really hate pinned issues.  Anyway, the fix will be tagged as stable later today.

jwnimmer-tri, 2024-01-09:
> My plan is not to investigate those until we have XCode 15 available in Drake CI.

As of recently (#20678), we have it available (and supported).

closed 2024-01-09 by jwnimmer-tri