RobotLocomotion/drake #20267: [simulator] The publish_every_time_step feature should be deprecated and removed

sherm1, 2023-09-27:
For historical reasons the Drake Simulator has an option to trigger Forced publish events every time step (and a separate option to do it during initialization). This was long ago superseded by PerStep events but a few uses remain.

In service of our ongoing work to rationalize the event system, it is time to nuke this anachronism.

jwnimmer-tri, 2025-12-19:
Part (2) is not required.  Drake maintainers will monitor the due date of deprecations and file the removal PR(s) ourselves.

I'll let @sherm1 reply to part (1).

sherm1, 2025-12-19:
Thanks for considering this, Khaled. In the two years since I posted this issue I'm not aware of them having caused any trouble. Consequently I'm less inclined to kill them off than I was then. Did they cause any particular trouble or confusion for you? If so then deprecation is probably wise. But if not, I'm inclined now to close this issue and just leave these in place. WDYT?