RobotLocomotion/drake #20268: Override the initial polytope used in IrisInConfigurationSpace

tomstewart89, 2023-09-27:
**Is your feature request related to a problem? Please describe.**
I'd like to implement the feature described in Section III E of the [IRIS-NP paper](https://arxiv.org/pdf/2303.14737.pdf); namely region refinement for runtime obstacles.

Specifically, I have a `HPolyhedron` that was returned from `IrisInConfigurationSpace`, I'd like to refine that polytope to account for the presence of a new task space collision object. To do that, I'd like to rerun `IrisInConfigurationSpace` with the updated task space collision geometry and pass in the current `HPolyhedron` as an initialization for the polytope.

Unfortunately there's currently no way to specify an initial polytope in `IrisInConfigurationSpace`, it's always initialized to the joint limits of the input `MultibodyPlant`.

**Describe the solution you'd like**
I'd like to be able to override the initial polytope used in `IrisInConfigurationSpace` with a user defined polytope. 

One option to do this would be to add a member: `std::optional<HPolyhedron> starting_polytope` to the `IrisOptions` struct. Then, if it's not a `std::nullopt`, it could be used to set P [here](https://github.com/RobotLocomotion/drake/blob/52d3fcf5b917cb1f0c75addf271d2807d27ab788/geometry/optimization/iris.cc#L464) rather than using the plant joint limits.

**Describe alternatives you've considered**
We could pass the `starting_polytope` as an argument to `IrisInConfigurationSpace` similar to the way that `const HPolyhedron& domain` is passed to the `Iris` function. This would require callers to specify the `starting_polytope` even if they don't plan to override it, which is pretty inconvenient. There's also already `starting_ellipse` member in `IrisOptions` so it feels natural to add a `starting_polytope` there too.

Unfortunately, since the `Iris` function also accepts an `IrisOptions` struct adding a `starting_polytope` member there introduces some ambiguity about whether P should be set [here](https://github.com/RobotLocomotion/drake/blob/52d3fcf5b917cb1f0c75addf271d2807d27ab788/geometry/optimization/iris.cc#L55C15-L55C16) to `domain` or `options.starting_polytope`. A simple solution to that would just be to `DRAKE_DEMAND` that `options.starting_polytope.has_value() == std::nullopt`, in the `Iris` function. I feel like there might be a better way to handle this problem though

sherm1, 2023-09-27:
Assigned to Russ for disposition.

tomstewart89, 2023-11-08:
Resolved with #20417

closed 2023-11-08 by tomstewart89