RobotLocomotion/drake #20294: Upgrade macOS to Python 3.12 to match Homebrew

jwnimmer-tri, 2023-10-02:
**Is your feature request related to a problem? Please describe.**

Hombrew is [adding Python 3.12 support](https://github.com/Homebrew/homebrew-core/pull/149142).  After it's landed and stabilized, they will switch hombrew's default Python on macOS to be 3.12, at which point Drake should also immediately switch to 3.12 on macOS.

**Describe the solution you'd like**

Prepare the build / CI / etc so that the switch is an easy flip.  See prior issue for details (TBD here).

**Additional context**

See #18252 for the prior issue (Python 3.11).

jwnimmer-tri, 2024-02-21:
The brew switch PR just landed (https://github.com/Homebrew/homebrew-core/pull/150390), so this ticket is ready to be worked now.

p-linnane, 2024-02-21:
Thanks for keeping an eye on our progress and sorry for the delay 🙇

MarcusSorealheis, 2024-02-22:
> so this ticket is ready to be worked now

🫡

svenevs, 2024-03-05:
Ok, the DEE component is fixed up and ready.  I intend to rebuild the images this evening after dinner.

@jwnimmer-tri my understanding is all the stamps are in place once I re-approve #21013.  So the plan of attack is

1. Have the images ready.
2. Switch Jenkins settings.
3. Merge #21013, continuous launches and results will come back.
4. Merge https://github.com/RobotLocomotion/drake-external-examples/pull/285

Only snag is if things go wrong in continuous for whatever reason, I won't be able to revert them before nightlies.

svenevs, 2024-03-06:
Alright, we're off to the races :horse_racing: Continuous jobs of interest:

- https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-ventura-clang-bazel-continuous-release/297/consoleFull
- https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-x86-monterey-clang-bazel-continuous-release/1575/consoleFull
- https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-sonoma-clang-bazel-continuous-release/224/consoleFull

If those and nightlies pass tomorrow morning, I think we can consider this complete.  Thanks for helping get things in motion @MarcusSorealheis!

svenevs, 2024-03-06:
>  Stable release and ship new wheels.

Whoops, sorry.  That's the last one, if nightlies pass then the image side should be good.

jwnimmer-tri, 2024-03-07:
I think we can call this finished.  We don't need to keep it open until the stable tag next week.

I checked the nightly build's wheel files and tgz files, and they all look correct.

closed 2024-03-07 by jwnimmer-tri