RobotLocomotion/drake #20296: [wheel] Use pymosek, removing libmosek from drake wheels

jwnimmer-tri, 2023-10-02:
**Is your feature request related to a problem? Please describe.**

Every few months, `pydrake` gets smaller as we optimize our build, but `libmosek.so` keeps getting bigger.  The huge file size of Mosek is pushing us ever closer to PyPI's hosting limits, and is an unnecessary download cost and security hazard for users who don't want to use Mosek.

**Describe the solution you'd like**

The beautiful https://github.com/yugr/Implib.so tool lets us do lazy _loading_ of shared libraries the first time they are called, instead of only lazy _binding_.  It seems like Mach-O might already have a similar feature built-in.

That should allow us to compile `libdrake.so` with load-time-unresolved references to Mosek, that are only resolved upon the first actual call into Mosek (which would be guarded by the MOSEKLM_LICENSE_FILE environment variable already).  For our wheel builds, we should able to resolve those loads against the copy of `libmosek.so` shipped in https://pypi.org/project/Mosek/, instead of shipping our own copy.

I will continue to investigate, time permitting.

**Additional context**

This would pave the way to do #10804 likewise.

jwnimmer-tri, 2025-07-21:
(1) Please change the manylinux wheels to drop two of the the three MOSEK license-related documents in the wheel file:
- drake_mosek_redistribution.txt -- keep intact in the wheel
- mosek-eula.pdf -- remove from the wheel
- LICENSE.third_party -- remove from the wheel

(2) Should we do the same for macOS?

The amount of slimming on manylinux was around 14%.  On macOS, a quick test with suggests only a 9% gain.  Still, that's probably worth at least trying.

Please take a spin at trying to achieve the same effect for macOS.  If we don't have a plausible prototype after investing 30 hours, we'll call it "not worth it" and give up.  I believe that budget has enough time to try to integration any existing capabilities (something built-in on macOS, or a different tool that implib).  I'm not interested in Drake hand-rolling an entirely new lazy loading tool from scratch; I want something "off the shelf".

mwoehlke-kitware, 2025-08-04:
Well... nuts. Symlinks appear to work. **If** I create them manually. It appears that `pip` explicitly does *not* extract symlinks.

jwnimmer-tri, 2025-08-04:
Hmm.  That makes me wonder why the chlpath wasn't working.  Is the current revision of #23243 the code that was segfaulting?  (If not, could you push that code?).  And then please fire off a wheel jenkins build so I can see the error.