RobotLocomotion/drake #20300: MultibodyPlant's discrete update should return EventStatus rather than throw

sherm1, 2023-10-02:
(See issue #20298 for more info)

When a step fails (say by failure to converge in TAMSI or SAP) MbP currently throws, leaving the state uncertain and recovery awkward for callers. It should instead return a "failed" EventStatus with an appropriate message embedded. Then the caller can decide whether to throw or recover.

Currently this applies only to MbP in discrete mode, since we don't have a similar way to return status from derivative calculations in a continuous simulation.

(Assigning initially to @amcastro-tri for disposition)

edrumwri, 2024-03-03:
It seems like "recovery" for discrete MbP failing to converge at a particular step size would require rebuilding the MbP with a smaller time step. IIRC there was discussion somewhere (slack? github issue?) about potentially altering MbP step size between simulation steps, but I'm failing to find it. Would it be feasible to temporarily reduce the MbP step size upon a convergence failure?

jwnimmer-tri, 2024-11-02:
I'm bumping up the priority on this somewhat.  Users of MbP in simulations are catching exceptions with string-pattern-matching hacks to notice when a step fails, and that's just too brittle.