RobotLocomotion/drake #20336: Use new bazel feature prefer_pic_for_opt_binaries

jwnimmer-tri, 2023-10-09:
**Is your feature request related to a problem? Please describe.**

The bazelrc flag `--force_pic` is problematic for users, since they _must not_ apply it on macOS but _should_ apply it on Ubuntu (for better build throughput).

See https://github.com/RobotLocomotion/drake/issues/20217 for some background.

**Describe the solution you'd like**

Once `prefer_pic_for_opt_binaries` is available (ETA as of Bazel 7.0), Drake should switch to use it, instead.

tyler-yankee, 2025-07-14:
Do drake_bazel_external users need to apply `prefer_pic_for_opt_binaries` for their own builds, or does it get baked in when Drake uses it?

jwnimmer-tri, 2025-10-29:
My PR finally merged upstream, so as of `rules_cc@0.2.14` we should be able to finish this.

They are releasing frequently these days in prep for Bazel 9, so hopefully it won't be too long.