RobotLocomotion/drake #20344: From point_cloud_test_parallel -- OMP: Error #15: Initializing libomp.so.5, but found unknown library already initialized.

rpoyner-tri, 2023-10-10:
### What happened?

Seen in https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-clang-bazel-continuous-everything-release/745/:
```
Using drake_cc_googletest_main.cc

[==========] Running 9 tests from 2 test suites.
[----------] Global test environment set-up.
[----------] 8 tests from PointCloudTest
[ RUN      ] PointCloudTest.TestExpectedNumThreads
OMP: Error #15: Initializing libomp.so.5, but found unknown library already initialized.
OMP: Hint This means that multiple copies of the OpenMP runtime have been linked into the program. That is dangerous, since it can degrade performance or cause incorrect results. The best thing to do is to ensure that only a single OpenMP runtime is linked into the process, e.g. by avoiding static linking of the OpenMP runtime in any library. As an unsafe, unsupported, undocumented workaround you can set the environment variable KMP_DUPLICATE_LIB_OK=TRUE to allow the program to continue to execute, but that may cause crashes or silently produce incorrect results. For more information, please see http://openmp.llvm.org/

//perception:point_cloud_test_parallel                                   FAILED in 0.6s

```
Not able to reproduce locally. @jwnimmer-tri thinks we have seen this once before, but I have not found the prior occurrence.

### Version

 3968c1c049d84390b4cba2223962a6e153c1ca76 

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
See above.
```

rpoyner-tri, 2023-10-10:
Prior occurrence(s) slack discussion: https://app.slack.com/client/T0JNB2DS4/C0KDGF4V9/thread/C0KDGF4V9-1675280469.824469

jwnimmer-tri, 2023-10-10:
My plan is to wait until #20251 is done and then see if this still matters.

BetsyMcPhail, 2024-02-15:
Another occurance on 2/15: [linux-focal-unprovisioned-clang-bazel-nightly-everything-release](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-unprovisioned-clang-bazel-nightly-everything-release/1367/)

jwnimmer-tri, 2024-03-19:
This is pretty rare.  I'll close until it becomes painful enough to necessitate mitigation.

closed 2024-03-19 by jwnimmer-tri

BetsyMcPhail, 2024-08-15:
This showed up again on 8/15 on Jammy --> https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-debug/177/

BetsyMcPhail, 2024-12-13:
12/13 - https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-release/1172

BetsyMcPhail, 2024-12-19:
12/19 - https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-release/1194/

ggould256, 2025-03-06:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-unprovisioned-clang-bazel-nightly-everything-debug/977/consoleFull

This doesn't seem to be restricted to point_cloud_test_parallel (in this case it was @common_robotics_utilities_internal//:parallelism_test ), so I'm also going to update the title.

xuchenhan-tri, 2025-03-17:
3/17: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-experimental-everything-release/7687

SeanCurtis-TRI, 2025-03-18:
3/18 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-everything-release/1524/

cohnt, 2025-03-24:
3/24 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-experimental-everything-release/7761/

Aiden2244, 2025-03-25:
3/25 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-release/1547/

tyler-yankee, 2025-04-07:
4/5: https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-release/412/

Aiden2244, 2025-04-18:
4/18 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-release/425/

jwnimmer-tri, 2025-05-07:
5/6 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-experimental-everything-release/8330/

ggould-tri, 2025-05-12:
https://drake-jenkins.csail.mit.edu/view/Production/job/linux-jammy-unprovisioned-clang-bazel-nightly-everything-release/1043/consoleFull

Aiden2244, 2025-06-06:
6/6/2025
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-release/1755/

tyler-yankee, 2025-07-16:
7/16 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-release/515/

tyler-yankee, 2025-12-31:
12/31 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-release/685/

It's been a while with this one until now. FWIW, this seems to be a Jammy-specific issue, so it probably just evaporates when Drake drops Jammy support next year.

BetsyMcPhail, 2026-03-12:
Jammy CI has been removed, closing