RobotLocomotion/drake #20350: Jenkins: Overprovisioned for Current Usage

svenevs, 2023-10-11:
While doing the Jenkins updates we noticed

1. Our instance type is overprovisioned.  The optimizer tool recommends using an arm64 host instead of x86, but since we backup and mount all of `$JENKINS_HOME`, we have concerns over possible conflicts dealing with compiled binary artifacts and java `.class` files floating around or something.  However, we can save substantially more (almost double) by switching to an arm64 host (provided we know how to do so safely, I haven't worked with Java in years).
2. We have overprovisioned how much space we are using for `$JENKINS_HOME`, probably safely get away with something significantly smaller.

    ```console
    $ df -H
    Filesystem      Size  Used Avail Use% Mounted on
    ...
    /dev/nvme1n1    1.1T   75G  928G   8% /var/lib/jenkins
    ```

The two of those combined could scrape back a decent chunk of change where AWS costs are concerned, but for (1) Betsy and I were worried about development time and unknown pitfalls by switching to arm64.

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2023-10-11:
As of #19099 survey a few months ago, the Jenkins master was $700 per month.  Even if we could save $400 per month, over the course of a year that's only like 20 hours of developer time.  Switching cost seems like it could easily spill over that number.

So, it doesn't seem to me like this is worth spending any time on.

Did I miss anything in my analysis?

svenevs, 2023-10-11:
Works for me.  @BetsyMcPhail may have additional thoughts, the EBS volume is the thing that's expensive here in particular.  We seem to be paying for almost 3x storage, PLUS whatever it costs to keep the listing of 1TB backusp daily for a month (drake-jenkins autobackups).

I didn't pull up the actual billing, but using the pricing calculator with some fast gp3 numbers matching our current instance, it's the difference between ~450/year for 250GBs, or ~1600/year for what we're doing.  It's not clear to me if the backups are included in that price.

Only filed the issue because this was originally going to surface in f2f and I didn't want it to be forgotten.  I would be absolutely thrilled to simply close this as #wontfix :upside_down_face:

jwnimmer-tri, 2023-10-11:
When Jenkins runs out of space, extremely bad things happen.  If our build logs start getting a lot bigger, it's nice to have room to grow.  Probably a 10x margin is more than we need, but 5x margin would be a low bar and we might be adding more builds in the future too (arm64 ubuntu).  So I'm inclined to leave well enough alone.

closed 2023-10-11 by jwnimmer-tri