RobotLocomotion/drake #20354: New Provisioned images for nasm

jwnimmer-tri, 2023-10-11:
Per https://github.com/RobotLocomotion/drake/pull/20353, we'll need new Provisioned images spun up with `nasm` also installed.

New images, in order of priority:
- [ ] Ubuntu Jammy
- [ ] Ubuntu Focal
- [ ] macOS Arm Ventura
- [ ] macOS x86_64 Monterey
- [ ] macOS Arm Monterey (could skip if you think we're turning this off soon)

(I think that's all of them?)

svenevs, 2023-10-16:
I'll be building these images out soon.

> macOS Arm Monterey (could skip if you think we're turning this off soon)

I can build monterey and ventura side by side on orka in the same amount of time.  I only need to take half of the ventura runner to do this so we should not experience interruptions, but the build queue may get a little long.

Now that the release is out, do I have a green light to do this within the next few days?

jwnimmer-tri, 2023-10-16:
> Now that the release is out, do I have a green light to do this within the next few days?

Yes: https://github.com/RobotLocomotion/drake/issues/20339#issuecomment-1765131498

Unless you think we're going to keep Arm Monterey around for more than say 2-3 weeks, you don't need to make a new image for it.  I don't intent to land the nasm-using pull request for several more weeks.

svenevs, 2023-10-22:
Images have been re-provisioned, nightlies will run with them tonight.

Linux images appear to be working exactly as desired:

- https://drake-jenkins.csail.mit.edu/job/linux-focal-gcc-bazel-experimental-release/12651/consoleFull
- https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-release/7044/consoleFull

Orka images appear to be working the same, the jobs are expected to finish successfully:

- https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-experimental-release/2921/consoleFull
- https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-clang-bazel-experimental-release/128/consoleFull

vSphere image appears to be working correctly, some long-running jobs from the bazel 6.4.0 pr:

- https://drake-jenkins.csail.mit.edu/job/mac-x86-monterey-clang-bazel-experimental-everything-release/22/consoleFull
- https://drake-jenkins.csail.mit.edu/job/mac-x86-monterey-clang-bazel-experimental-release/83/consoleFull

-----

Linux provisioning notes to CC @BetsyMcPhail on:

- drake-linux-jammy-2023-10-05

    ```
    root@ip-172-31-7-228:~# apt-get upgrade                                            
    Reading package lists... Done                                                      
    Building dependency tree... Done                                                   
    Reading state information... Done                                                  
    Calculating upgrade... Done                                                        
    The following packages have been kept back:                                        
      libegl-mesa0 libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 linux-aws        
      linux-headers-aws linux-image-aws                                                
    The following packages will be upgraded:                                           
      cloud-init curl irqbalance libcurl3-gnutls libcurl4 libcurl4-openssl-dev         
      libnetplan0 libsgutils2-2 libtiff-dev libtiff5 libtiffxx5 linux-libc-dev         
      netplan.io sg3-utils sg3-utils-udev sosreport vim vim-common vim-runtime         
      vim-tiny xxd
    ```

- drake-linux-focal-2023-10-05

    ```
    root@ip-172-31-23-13:~# apt-get upgrade                                            
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Calculating upgrade... Done
    The following packages have been kept back:
      linux-aws linux-headers-aws linux-image-aws
    The following packages will be upgraded:
      cloud-init curl libcurl3-gnutls libcurl4 libtiff-dev libtiff5 libtiffxx5
      linux-libc-dev sosreport vim vim-common vim-runtime vim-tiny xxd
    14 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
    ```

I did **not** upgrade the kernel for either focal or jammy, we skipped the last unprovisioned kernel upgrade due to lack of availability.  If upgrading the kernel image for provisioned is something we want to do, I can fold that into the bazel 6.4 update.

-----

I went ahead and reprovisioned now since for bazel 6.4 we only need to redo linux (because of `bazelisk`).  We can also close this ticket and turn the last checkbox of "clean up the old images next week" into its own issue if we prefer :+1:

jwnimmer-tri, 2023-10-24:
Yeah, any "regular AMI package upgrade" maintenance task should be its own ticket, not this one.

closed 2023-10-24 by jwnimmer-tri