RobotLocomotion/drake #20360: Hard to find the source of an illlustration phong-diffuse warning message

ggould-tri, 2023-10-12:
We have a warn-once about setting phong diffuse color in an illustration geometry at https://github.com/RobotLocomotion/drake/blob/905fc397d442334d5626bbc7493f493c722b5f22/geometry/geometry_state.cc#L1102
```
  if (properties.HasProperty("phong", "diffuse_map")) {
    static logging::Warn log_once(
        "Explicitly defined values for the ('phong', 'diffuse_map') property "
        "are not currently used in illustration roles -- only perception "
        "roles");
  }
```

If you get this warning in your build, however, it's nearly impossible to find the actual error.  The actual AssignRole call is dozens of stack frames down the Drake parser stack.

**Is your feature request related to a problem? Please describe.**
I spent three hours chasing one of these warnings this morning, and got nowhere.

**Describe the solution you'd like**
The message should provide hints about the model and/or file involved and other such breadcrumbs.

**Describe alternatives you've considered**
Upgrading the warning to a throw does not result in a usable stack trace.

ggould-tri, 2023-10-16:
I'm just trying to reduce log spam downstream.  We've got a whole lot of random warnings that make debugging more confusing, and I thought (incorrectly) that this would be an easy one to fix because it was just some bad content in one of our model files (which maybe it is) that I could find and fix (nope).

IMO if there is a validity constraint on a file or a call, we should detect it and log it as early as possible.  Logging this far down the call stack isn't especially helpful to anyone.  How is a user supposed to act on this message?

Reducing it to debug would probably solve my problem, so I wouldn't object to that, but it seems like the wrong approach if this is a real validity concern.  If it's meant to be actionable to a user (e.g. by fixing the relevant urdf/sdf/callsite/whatever) it needs to be caught while there's still enough context to give a meaningful error; if it's not meant to be user-actionable, a debug message seems no better than a TODO to a relevant issue.

jwnimmer-tri, 2024-07-09:
Fixed in #21633.

closed 2024-07-09 by jwnimmer-tri