RobotLocomotion/drake #20372: IrisFromCliqueCover

AlexandreAmice, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
Implement the algorithm outlined in [Approximating Robot Configuration Spaces with few Convex Sets
using Clique Covers of Visibility Graphs](https://arxiv.org/pdf/2310.02875.pdf) for automatically decomposing configuration spaces using IRIS.

**Describe the solution you'd like**
The algorithm has 5 steps:

- [ ] Sample configuration space points 
- [x] Construct Visibility Graph (implemented in [VisibilityGraph](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1planning.html#adca2fc53de41fd4760fe9825e079456e))
- [ ] Solve Clique Cover
- [x] Compute Outer Lowner John Ellipsoid ([implemented in [MinimumVolumeCircumscribedEllipsoid](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1optimization_1_1_hyperellipsoid.html#a0783f7f2e02468471d114e49cbbfa4fd))
- [x] Inflate Polytopes. (can be achieved by running the appropriate variant of IRIS with the `iteration_limit` set to 1)

**Additional context**
@RussTedrake has a draft PR of this laid out in #20013.
@wernerpe for discussion.

wernerpe, 2023-10-18:
From a practical point of view, it seems to me that we would want the user to switch between two cases:
1. Clip the eigen values at some large value which prevents numerical issues. This means  that the ellipsoids cannot get infinitely skinny.
2. Explicitly remove the clipping and return an affine ball instead, that allows for lower-dimensional ellipsoids. (Maybe by having the user specify max_eigenval = inf ?)

The hyperellipsoid can represent infinitely large expansions, the affine ball can represent lower-dimensional ellipsoids in a higher-dimensional space. It seems to me we cannot have both at the same time. Option 1 is what would be practical for VCC.

AlexandreAmice, 2023-11-13:
Solve Clique Cover step is now handled by #20401

AlexandreAmice, 2024-02-23:
Closed by #20831

closed 2024-02-23 by AlexandreAmice