RobotLocomotion/drake #20376: `pydrake.tutorials` does not work due to Jupyter notebook v7

Harimus, 2023-10-17:
### What happened?

After trying to install pydrake ( with `pip install drake`)  on my local Ubuntu 20.04, I couldn't make the `python3 -m pydrake.tutorials` command work, even after installing `notebook`.

After checking through the script and online, it seems like the cause was that in the script,
```from notebook import notebookapp as app```
errors out with `ModuleNotFoundError`, because `pip install notebook` now installs notebook version 7.0.5 by default. [downgrading it to notebook==6.5.5](https://github.com/microsoft/azuredatastudio/issues/23945#issuecomment-1655543692) fixed this issue.

The default notebook version installed by `sudo apt-get install jupyter-notebook` seem to be `6.0.5`, so on the system python3.8 this shouldn't happen. 

But since the documentation recommend setting up `venv` for python (pip) install, the solution suggested in the script, `pip install notebook` seems to point towards the wrong version.

maybe the `__doc__` string in `tutorials.py` could be updated to a fixed version or updated to work on notebook v7?

### Version

drake==1.22.0

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
ERROR: the Jupyter notebook runtime is not installed!

Drake offers Python tutorials that can be previewed and executed as Jupyter
notebooks online with no need for local installation. To run Drake's tutorials
online, refer to the `Drake Tutorials <https://drake.mit.edu/>`_ website.

Alternatively, to run Drake's tutorials locally from an
`installed <https://drake.mit.edu/installation.html>`_ copy of Drake,
run ``python3 -m pydrake.tutorials`` to launch a Jupyter browser.

Be sure your ``PYTHONPATH`` has been set per the installation instructions,
e.g., via ``source env/bin/activate`` in the
`pip instructions <https://drake.mit.edu/pip.html>`_.

If you haven't done so already, you'll also need to install the Jupyter
notebook package on your system:

- For pip or macOS, use: ``pip install notebook``.
- For Ubuntu, use ``sudo apt-get install jupyter-notebook``.
```

jwnimmer-tri, 2023-10-17:
Thanks for reporting.

I expect the fix is going to be to change the https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/tutorials.py launcher to work with either module structure of Jupyter, via some kind of try-again or auto-detection.

jwnimmer-tri, 2023-10-19:
Our goal is for Drake to work with both notebook <= 6 and notebook >= 7, with the same code.  Thus, we need to insert something like `try <import guess 1> except ImportError: <import guess 2>` into our loader so that it will gracefully fall back no matter which version the user has.

Harimus, 2023-10-20:
I'm not familiar with the inner workings of JupyterLab/notebooks, but a quick look at version 7.0.5 seems to indicate the existence of `notebook.app` so replacing the 
``` rom notebook import notebook as app```
to 
```from notebook import app```
should fix this I think? (I tried to change that line on my local env with version 7 and it seems to work)

So something like this?

```python
import importlib.metadata as metadata                                      
try:
    if int(metadata.version('notebook').split('.')[0]) < 7:                          
        from notebook import notebookapp as app                             
    else:                                                                   
        from notebook import app        
except metadata.PackageNotFoundError:
    ........... #same as before
```
(or alternatively just `import notebook. notebook.__version__` instead of `importlib.metadata`

jwnimmer-tri, 2023-10-20:
My proposed fix is #20408.  In my local testing, both 6.4 and 7.0 work OK now, but please let me know if you still see problems.

closed 2023-10-23 by EricCousineau-TRI