RobotLocomotion/drake #20377: LoadYamlString on variants with defaults does not respect the defaults

dmcconachie-tri, 2023-10-17:
### What happened?

When using YAML for scenarios with overrides, if there is a variant in the scenario the overrides can either fail to load entirely or end up with a defaulted version of the variant rather than the expected contents of the scenario file depending on how one formats the override text. Here's a minimal reproduction of both versions of the parse issue:

https://github.com/dmcconachie-tri/drake/tree/yaml_variant_override

### Version

e8e8a32374c6eda19af3885b83284fb945482c5f

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
==================== Test output for //common/yaml:yaml_io_test:
Using drake_cc_googletest_main.cc

[==========] Running 17 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 17 tests from YamlIoTest
[ RUN      ] YamlIoTest.LoadString
[       OK ] YamlIoTest.LoadString (0 ms)
[ RUN      ] YamlIoTest.LoadStringChildName
[       OK ] YamlIoTest.LoadStringChildName (0 ms)
[ RUN      ] YamlIoTest.LoadStringDefaults
[       OK ] YamlIoTest.LoadStringDefaults (0 ms)
[ RUN      ] YamlIoTest.LoadStringOptions
[       OK ] YamlIoTest.LoadStringOptions (0 ms)
[ RUN      ] YamlIoTest.LoadStringDefaultsAndOptions
[       OK ] YamlIoTest.LoadStringDefaultsAndOptions (0 ms)
[ RUN      ] YamlIoTest.LoadFile
[       OK ] YamlIoTest.LoadFile (0 ms)
[ RUN      ] YamlIoTest.LoadFileChildName
[       OK ] YamlIoTest.LoadFileChildName (0 ms)
[ RUN      ] YamlIoTest.LoadFileWithVariantNonDefault
unknown file: Failure
C++ exception with description "<string>:3:3: YAML node of type Mapping (with size 1 and keys {some_int32}) key 'some_int32' did not match any visited value entry for <root> while accepting YAML node of type Mapping (with size 1 and keys {value}) while visiting drake::yaml::test::DoubleStruct value." thrown in the test body.

[  FAILED  ] YamlIoTest.LoadFileWithVariantNonDefault (1 ms)
[ RUN      ] YamlIoTest.LoadFileWithVariantNonDefaultWithTag
common/yaml/test/yaml_io_test.cc:169: Failure
Expected equality of these values:
  value.some_bool
    Which is: false
  true

[  FAILED  ] YamlIoTest.LoadFileWithVariantNonDefaultWithTag (0 ms)
[ RUN      ] YamlIoTest.LoadFileChildNameBad
[       OK ] YamlIoTest.LoadFileChildNameBad (0 ms)
[ RUN      ] YamlIoTest.LoadFileSchemaBad
[       OK ] YamlIoTest.LoadFileSchemaBad (0 ms)
[ RUN      ] YamlIoTest.SaveString
[       OK ] YamlIoTest.SaveString (0 ms)
[ RUN      ] YamlIoTest.SaveStringChild
[       OK ] YamlIoTest.SaveStringChild (0 ms)
[ RUN      ] YamlIoTest.SaveStringDefaults
[       OK ] YamlIoTest.SaveStringDefaults (0 ms)
[ RUN      ] YamlIoTest.SaveFile
[       OK ] YamlIoTest.SaveFile (0 ms)
[ RUN      ] YamlIoTest.SaveFileAllArgs
[       OK ] YamlIoTest.SaveFileAllArgs (0 ms)
[ RUN      ] YamlIoTest.SaveFileBad
[       OK ] YamlIoTest.SaveFileBad (0 ms)
[----------] 17 tests from YamlIoTest (3 ms total)

[----------] Global test environment tear-down
[==========] 17 tests from 1 test suite ran. (3 ms total)
[  PASSED  ] 15 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] YamlIoTest.LoadFileWithVariantNonDefault
[  FAILED  ] YamlIoTest.LoadFileWithVariantNonDefaultWithTag

 2 FAILED TESTS
================================================================================
```

jwnimmer-tri, 2023-10-17:
This is working as intended.  Any time a value for a variant is given, you must also write down the type tag (unless it's the first type, in which case we assume that answer).  The absence of a tag does not denote "leave the type intact", it denotes "select the first type of the variant".

dmcconachie-tri, 2023-10-17:
Okay, then one of the versions of this is no problem, but the second example is still a bug; when using overrides, we lose the value that we passed in as `default` and instead get a default constructed result.

jwnimmer-tri, 2023-10-17:
Ah, I see.

Sometimes the existing defaults are nuked (for scalars and arrays) and sometimes they are merged (for maps and structs).  I guess we never really specified where "variant of structs" falls in that regime.  I can understand the assumption being that they would merge, since struct merge.  I think you're probably right that we should specify that variant-of-structs uses merging.

PR #20391 (2023-10-19): [yaml] Retain default values when parsing a variant
Closes #20377.

+@dmcconachie-tri for feature review, please -- or at least to check that the test expectations are as-desired.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20391)
closed 2023-10-19 by EricCousineau-TRI