RobotLocomotion/drake #20425: FlattenModelDirectives does not do scoping

ggould-tri, 2023-10-25:
### What happened?

This is a longstanding problem that has only just become urgent for me; I'm creating this issue to associate with my work to fix it.

`FlattenModelDirectives` has an implementation comment `// NOTE: Does not handle scoping!`.  This is accurate:  Flattened model directives lack the scoping requested by the original directives, and so are typically invalid for any model with cross-references or duplicate models.

It also has no documentation comment, nor is it tested.

In fact, to the extent that `FlattenModelDirectives` is useful (to make a durable, self-contained record of a scenario-as-executed; to create non-colliding content for hashing), it must be the case that processing model directives in deep form and flattened form should result in the same model instances.  As written it does not achieve that.

Fixing this probably also entails making the assignment of scoped names more explicit.

ggould-tri, 2023-10-25:
FYI @IanTheEngineer

jwnimmer-tri, 2023-10-31:
I agree that `FlattenModelDirectives` was broken because it ignored the namespaces.

> ... to make a durable, self-contained record of a scenario-as-executed ...

Even after calling `FlattenModelDirectives`, the scenario is not self-contained.  It will still have dmd elements like `add_model: { file: package://..., ... }` which refer to other files, and those other files can still refer to other files, with no upper bound in the inclusion depth.

So specifically: the only benefits of `FlattenModelDirectives` are that (1) it slightly reduces the number of files that must be archived (or hashed) to capture a repeatable scenario specification, and (2) it provides for a slightly more compact summary, which might be easier for humans to grok.

ggould-tri, 2023-10-31:
Yes, it is correct that flattening model directives only solves the marshalling problem as far as the model directives format itself; marshalling the closure of the things it references is a separate problem not addressed yet.

PR #20438 (2023-11-10): Correct, test, and document `FlattenModelDirectives`
FlattenModelDirectives did not handle scoped names, which made it difficult to use in scenarios with multiple instances of the same model.

It was also undocumented and untested.

This change adds correct namespace handling; it also adds documentation and a minimal test (minimal because any naming-related errors are caught during the assembly of the MultibodyTree, and because the scoped test fixtures were designed to exercise naming).

 * Fixes #20425.
 * Removes misleading comments (including a mention of RigidBodyTree!).
 * Documents Flatt [... cut]
commit be76e976 (2023-11-10): Correct, test, and document FlattenModelDirectives (#20438)

FlattenModelDirectives did not handle scoped names, which made it difficult to use in scenarios with multiple instances of the same model.

It was also undocumented and untested.

This change adds correct namespace handling; it also adds documentation and a minimal test (minimal because any naming-related errors are caught during the assembly of the MultibodyTree, and because the scoped test fixtures were designed to exercise naming).

 * Fixes #20425.
 * Removes misleading comments (including a mention of RigidBodyTree!).
 * Documen [... cut]
closed 2023-11-10 by jwnimmer-tri