RobotLocomotion/drake #20436: De-commission macOS x86 debug & asan builds

jwnimmer-tri, 2023-10-26:
In the same spirit as #20339, we are trimming down our Monterey support matrix, hopefully transition it to GitHub Actions eventually.  Therefore, we should stop running the really long / expensive builds -- anything Debug or ASan.

On second look, I think actually we're not running them anymore (they are not in Production).  So the request here is actually to just delete the macOS x86 Experimental builds that have no mate anywhere in Production.

mwoehlke-kitware, 2023-12-13:
There is still [mac-x86-monterey-clang-bazel-nightly-everything-debug](https://drake-jenkins.csail.mit.edu/view/Production/job/mac-x86-monterey-clang-bazel-nightly-everything-debug/). Removing not-everything-debug is easy. Removing everything-debug *except* for mac-x86-monterey-clang-bazel is... less trivial. Thoughts? (Maybe we want to remove that nightly build also?)

jwnimmer-tri, 2023-12-13:
OK here's a more specific goal ...

Keep these builds unchanged:
- mac-x86-monterey-clang-bazel-continuous-release
- mac-x86-monterey-clang-bazel-experimental-everything-release
- mac-x86-monterey-clang-bazel-experimental-release
- mac-x86-monterey-unprovisioned-clang-bazel-experimental-everything-release
- mac-x86-monterey-unprovisioned-clang-bazel-experimental-packaging
- mac-x86-monterey-unprovisioned-clang-bazel-experimental-release
- mac-x86-monterey-unprovisioned-clang-bazel-nightly-packaging
- mac-x86-monterey-unprovisioned-clang-bazel-nightly-release
- mac-x86-monterey-unprovisioned-clang-bazel-staging-packaging
- mac-x86-monterey-unprovisioned-clang-wheel-nightly-release
- mac-x86-monterey-unprovisioned-clang-wheel-staging-release
- mac-x86-monterey-vsphere-keepalive

Add this build:
- mac-x86-monterey-clang-bazel-nightly-everything-release

Delete these builds:
- mac-x86-monterey-clang-bazel-nightly-everything-debug

Ideally, delete these builds (because they have no Production counterpart):
- mac-x86-monterey-clang-bazel-experimental-address-sanitizer
- mac-x86-monterey-clang-bazel-experimental-debug
- mac-x86-monterey-clang-bazel-experimental-everything-address-sanitizer
- mac-x86-monterey-clang-bazel-experimental-everything-debug
- mac-x86-monterey-clang-cmake-experimental-everything-release
- mac-x86-monterey-clang-cmake-experimental-release
- mac-x86-monterey-unprovisioned-clang-cmake-experimental-everything-release
- mac-x86-monterey-unprovisioned-clang-cmake-experimental-release

If it's somehow really annoying to special case the prior list, I guess it's OK if they stay.

BetsyMcPhail, 2023-12-14:
All Mac ASan builds were removed in https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/104, as part of [#20339](https://github.com/RobotLocomotion/drake/issues/20339)

mwoehlke-kitware, 2023-12-14:
In general (but particularly for macOS), do we *ever* need a build that is only experimental? Or should every experimental build have at least a non-production scheduled build?

jwnimmer-tri, 2023-12-14:
(1) All continuous/nightly/weekly jobs should always have an experimental twin.

(2) ASan and Debug are _actively unsupported_ on macOS x86 (because they certainly fill up the disk and bomb out) so we should not even offer them as options.  They are known to fail.

(3) For other build variations that are not in production in their exact variant sometimes we want to "fill in" the experimental matrix in support of prototyping.  Because they are "within the footprint" of other vectors in the config space, they are expected to pass.

BetsyMcPhail, 2023-12-15:
`mac-x86-monterey-clang-bazel-nightly-everything-debug` has been replaced with `mac-x86-monterey-clang-bazel-nightly-everything-release`

svenevs, 2023-12-20:
As of today, these are the jobs we have remaining:

```
curl https://drake-jenkins.csail.mit.edu/api/json?tree=jobs[name] | jq '.jobs[].name' | /bin/grep -i x86
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 44579    0 44579    0     0   186k      0 --:--:-- --:--:-- --:--:--  186k
"mac-x86-monterey-clang-bazel-continuous-release"
"mac-x86-monterey-clang-bazel-experimental-everything-release"
"mac-x86-monterey-clang-bazel-experimental-release"
"mac-x86-monterey-clang-bazel-nightly-everything-release"
"mac-x86-monterey-clang-cmake-experimental-everything-release"
"mac-x86-monterey-clang-cmake-experimental-release"
"mac-x86-monterey-unprovisioned-clang-bazel-experimental-everything-release"
"mac-x86-monterey-unprovisioned-clang-bazel-experimental-packaging"
"mac-x86-monterey-unprovisioned-clang-bazel-experimental-release"
"mac-x86-monterey-unprovisioned-clang-bazel-nightly-packaging"
"mac-x86-monterey-unprovisioned-clang-bazel-nightly-release"
"mac-x86-monterey-unprovisioned-clang-bazel-staging-packaging"
"mac-x86-monterey-unprovisioned-clang-cmake-experimental-everything-release"
"mac-x86-monterey-unprovisioned-clang-cmake-experimental-release"
"mac-x86-monterey-unprovisioned-clang-wheel-experimental-release"
"mac-x86-monterey-unprovisioned-clang-wheel-nightly-release"
"mac-x86-monterey-unprovisioned-clang-wheel-staging-release"
"mac-x86-monterey-vsphere-keepalive"
```

mac x86 debug and asan jobs are officially gone :tada:

BetsyMcPhail, 2023-12-20:
@mwoehlke-kitware from the list above, the following experimental jobs still should be removed:

- mac-x86-monterey-clang-cmake-experimental-everything-release
- mac-x86-monterey-clang-cmake-experimental-release
- mac-x86-monterey-unprovisioned-clang-cmake-experimental-everything-release
- mac-x86-monterey-unprovisioned-clang-cmake-experimental-release

svenevs, 2024-01-03:
The four experimental jobs above have been removed.  What remains:

- Continuous:
    - `mac-x86-monterey-clang-bazel-continuous-release`
- Nightly:
    - `mac-x86-monterey-clang-bazel-nightly-everything-release`
    - `mac-x86-monterey-unprovisioned-clang-bazel-nightly-packaging`
    - `mac-x86-monterey-unprovisioned-clang-bazel-nightly-release`
    - `mac-x86-monterey-unprovisioned-clang-wheel-nightly-release`
- Staging:
    - `mac-x86-monterey-unprovisioned-clang-bazel-staging-packaging`
    - `mac-x86-monterey-unprovisioned-clang-wheel-staging-release`
- Experimental:
    - `mac-x86-monterey-clang-bazel-experimental-everything-release`
    - `mac-x86-monterey-clang-bazel-experimental-release`
    - `mac-x86-monterey-unprovisioned-clang-bazel-experimental-everything-release`
    - `mac-x86-monterey-unprovisioned-clang-bazel-experimental-packaging`
    - `mac-x86-monterey-unprovisioned-clang-bazel-experimental-release`
    - `mac-x86-monterey-unprovisioned-clang-wheel-experimental-release`
- Misc:
    - `mac-x86-monterey-vsphere-keepalive`