RobotLocomotion/drake #20439: Create process for regular Jenkins updates and maintenance

BetsyMcPhail, 2023-10-26:
Follow up to #18597. To ensure that Jenkins does not become far out of date again, we should setup a playbook for routine updates.

jwnimmer-tri, 2024-01-18:
For a pace here: Ideally, we would do monthly, unless that's a crazy amount of work?  If we can do it in like an hour or so, then monthly sounds great!  Maybe the first Friday or every month or something.

jwnimmer-tri, 2024-01-18:
Running `apt update && apt upgrade` every month is fine.

In terms of `dist-upgrade`, we definitely don't want to lose security updates so we can't get _too_ far behind, but there's no urgent need to immediately upgrade to 24.04 this spring.

svenevs, 2024-01-19:
Slack's `/remind` feature is limited, I can't do "first Friday of every month", so we will be reminded on the first of every month.

I also set some reminders for distribution upgrades for April 1st in 2025, 2027, 2029, and 2031.

Update instructions are documented in the drake-jenkins doc.

closed 2024-01-19 by svenevs