RobotLocomotion/drake #20444: [MuJoCo parser] Support parsing models from mujoco menagerie

RussTedrake, 2023-10-29:
The MuJoCo XML parser loads and properly displays all of the models in the DeepMind Control Suite (or will after my upcoming set of PRs) except for [dog.xml](https://github.com/google-deepmind/dm_control/blob/4d356d350b4c98921a92252bb58fc70df98cfa5b/dm_control/suite/dog.xml), which is an outlier with tendons and skin, etc. I've opened this issue to keep track of the (sad) state of our current MuJoCo parser in loading the [menagerie](https://github.com/google-deepmind/mujoco_menagerie) assets.

- `error: mesh * was defined with a non-uniform scale; but Drake currently only supports uniform scaling`.
  - `agility_cassie/cassie.xml`
  - `shadow_hand/left_hand.xml` and `right_hand.xml`
- Error:`Spatial inertia fails SpatialInertia::IsPhysicallyValid().` This may be that the models are actually wrong, but I hope that it's a problem with the parser! 
  - `anybotics_anymal_b/anymal_b.xml` and `anymal_c`
  - `franka_emika_panda/panda.xml` and `panda_hand.xml` (this file also fails to load some .stl files; if it was depending the volume of the stl mesh for inertia, that could explain the error)
  - `hello_robot_stretch/stretch.xml`
  - `realsense_d435i/d435i.xml`
  - `rethink_robotics_sawyer/sawyer.xml`
  - `universal_robots_ur10e/ur10e.xml`
  - `universal_robots_ur5e/ur5e.xml`
- OBJ assets, but textures are specified via `texture`, instead of .mtl.
  - `skydio_x2/x2.xml`
  - `unitree_a1/a1.xml`
  - `unitree_go2/go2.xml`
- Unfortunately many (most?) of the other models use STL assets; which Drake does not support

To reproduce, run e.g.
```
from pydrake.all import ModelVisualizer, StartMeshcat, PackageMap

visualizer = ModelVisualizer(meshcat=meshcat)
visualizer.parser().package_map().AddRemote(
    package_name="mujoco_menagerie",
    params=PackageMap.RemoteParams(
        # This repository doesn't have up-to-date tags/releases; the scary
        # hash in the url is the most recent commit sha at the time of my
        # writing.
        urls=[
            f"https://github.com/google-deepmind/mujoco_menagerie/archive/0c8c9315506dbd4e9b3c1a6ff6faa28612792d1d.tar.gz"
        ],
        sha256=(
            "ea7e63453236c1e8583b3dc7acc75d9204ef8b9b8407d75c2c6a0b75e0be202a"
        ),
        strip_prefix="mujoco_menagerie-0c8c9315506dbd4e9b3c1a6ff6faa28612792d1d/",
    ),
)
visualizer.AddModels(
    url="package://mujoco_menagerie/unitree_a1/a1.xml"
)
visualizer.Run(loop_once=True)
```

RussTedrake, 2024-06-29:
See this (TRI internal) [slack discussion](the legacy, non-standard, MuJoCo texture projection) for discussion about the extent to which we want to support "the legacy, non-standard, MuJoCo texture projection".

RussTedrake, 2024-07-04:
#21666 resolves some errors. I've just done a complete sweep through the menagerie and made a detailed list of errors/warnings above.

RussTedrake, 2024-09-12:
#21889 allowed meshcat to properly view a number of models that were not displayed properly before: boston_dynamics_spot, franka_emika_panda, franka_fr3, rethink_robotics_sawyer, shadow_hand/right_hand, unitree_go2, universal_robots_ur10e, and universal_robots_ur5e. Progress! I've updated the list.

RussTedrake, 2025-01-02:
fwiw, after #22370 lands, I have fixes for the two simple error queued up in my [`mujoco_more_than_one_orientation_fix`](https://github.com/RussTedrake/drake/commit/0fe66b3e4b671ffd3c436fa9e63bce3d926f6221) and [`mujoco_capsule_size`](https://github.com/RussTedrake/drake/commit/c81563d29fc35c87b878eb7a2eab1f8b98bed42c)  branches.

RussTedrake, 2025-01-06:
OK, I've addressed all of the small errors I know about. I've also create [`make_drake_compatible_model`](https://manipulation.mit.edu/python/make_drake_compatible_model.html) that will preprocess an xml (or urdf) to overcome all of the known errors.

A few more things on my list:

- [ ] Advertise `make_drake_compatible_model` from the parser.  Is it ok to advertise the manipulation repo version? Or is migrating this to drake a prerequisite? 
- [ ] Escalate warnings for missing meshes to errors.  It results in incorrect dynamics.  And now that `make_drake_compatible_model` is in better shape, there is a reasonable work-around.
- [ ] Capture warnings about missing textures; make sure they are representative of the experience in meshcat, at least.
- [ ] Resolve missing textures... possibly in `make_drake_compatible_model`, if's it's not currently viable in the parser. I think the immediate work-around is to generate an .mtl for the .obj that gets generated.  (or gltf, if that is now to be preferred)

@jwnimmer-tri , @rpoyner-tri -- would appreciate any feedback you have on these.

jwnimmer-tri, 2025-01-06:
> Advertise make_drake_compatible_model from the parser.

One nice tactic for situations like this would be have the parser's message hyperlink to a (new) anchor on https://drake.mit.edu/troubleshooting.html and then have that documentation explain in more detail (and link to work-arounds like that tool).  We don't want the parser message to be _too_ brief (it should still be self-contained and directly explain the problem), but IMO explaining work-arounds is best left to the website.

Or alternatively, if there is a specific issue filed we can link to the issue instead of the troubleshooting, and the issue OP can contain an explanation of the workarounds, and people can subscribe to the issue to track progress.

jwnimmer-tri, 2025-01-14:
> ... would appreciate any feedback you have on these.

>  Escalate warnings for missing meshes to errors. It results in incorrect dynamics.

SGTM

> Capture warnings about missing textures in the detail_mujoco_parser_examples_test; ...

No opinion.

> Resolve missing textures ...

I don't understand enough of the problem to offer an opinion.

RussTedrake, 2025-01-16:
I've added mujoco texture support to [`make_drake_compatible_model`](https://manipulation.mit.edu/python/make_drake_compatible_model.html) in the manipulation repo. [This example notebook](https://github.com/RussTedrake/manipulation/blob/c301864d3e95596cb53940e0784db6b71a33fece/book/robot/menagerie.ipynb) loads all of the robot models from the menagerie and visualizes them in meshcat. Apart from a few missing textures on a few model elements, and a few funky normals/windings picked up during the mesh file conversions, everything model now works or has a specific Drake issue tracking it.

commit 15eeeacf (2025-01-17): [mujoco parser] upgrade unsupported mesh to error

Previously if a mesh was unsupported, then we only produceed a warning
and ignored the associated geometry. Now it is an error. (We can use
make_drake_compatible_mesh in the manipulation repo to batch convert
stl to obj, etc).

Also removes the "if you said stl but I can find a juxtaposed obj,
I'll use it" logic.

The non-uniform mesh warning was turned to an error (non-uniform
meshes already threw an error, but later in the pipeline).

All of the errors now reference
http://drake.mit.edu/troubleshooting.html which points to
make_drake_compati [... cut]
PR #22481 (2025-01-18): [mujoco parser] upgrade unsupported mesh to error
Previously if a mesh was unsupported, then we only produceed a warning and ignored the associated geometry. Now it is an error. (We can use make_drake_compatible_mesh in the manipulation repo to batch convert stl to obj, etc).

Also removes the "if you said stl but I can find a juxtaposed obj, I'll use it" logic.

The non-uniform mesh warning was turned to an error (non-uniform meshes already threw an error, but later in the pipeline).

All of the errors now reference
http://drake.mit.edu/troubleshooting.html which points to make_drake_compatib [... cut]
commit 43820615 (2025-01-18): [mujoco parser] upgrade unsupported mesh to error (#22481)

Previously if a mesh was unsupported, then we only produceed a warning
and ignored the associated geometry. Now it is an error. (We can use
make_drake_compatible_mesh in the manipulation repo to batch convert
stl to obj, etc).

Also removes the "if you said stl but I can find a juxtaposed obj,
I'll use it" logic.

The non-uniform mesh warning was turned to an error (non-uniform
meshes already threw an error, but later in the pipeline).

All of the errors now reference
http://drake.mit.edu/troubleshooting.html which points to
make_drak [... cut]
closed 2025-01-18 by sherm1