RobotLocomotion/drake #20446: symbolic::Expression::Evaluate() doesn't take EqualToDict

hongkai-dai, 2023-10-29:
### What happened?

We recommend the users to use EqualToDict when constructing a map with symbolic::Variable as the key https://github.com/RobotLocomotion/drake/blob/c7289e0c0c90bbde4b98aef13daed38f0f6bdb1a/bindings/pydrake/symbolic/symbolic_py_monolith.cc#L624-L628, but when we call symbolic::Expression::Evaluate() with a EqualToDict, it fails. Here is the code to reproduce it
```py
import pydrake.symbolic as sym
from pydrake.common.containers import EqualToDict
x = sym.Variable("x")
expr = x ** 2
env = EqualToDict()
env[x] = 1
expr.Evaluate(env)
```
and I got the error
```

TypeError: Evaluate(): incompatible function arguments. The following argument types are supported:
    1. (self: pydrake.symbolic.Expression, env: Dict[pydrake.symbolic.Variable, float] = {}, generator: pydrake.common._module_py.RandomGenerator = None) -> float
    2. (self: pydrake.symbolic.Expression, generator: pydrake.common._module_py.RandomGenerator) -> float

Invoked with: <Expression "pow(x(0), 2)">, {<pydrake.common.containers.EqualToDict.__init__.<locals>.Proxy object at 0x7f7036de38b0>: 1}

```

Should we support `EqualToDict` in `symbolic::Expression::Evaluate()`?

BTW, the reason why I try to use `EqualToDict` instead of `dict`, was that I got the error when I use `dict` to map s symmetric symbolic matrix to its values
```py
In [21]: prog = solvers.MathematicalProgram()

In [22]: X = prog.NewSymmetricContinuousVariables(3, "X")

In [23]: X_val = np.array([[1, 2, 3], [4, 5, 6], [7, 8, 9.]])

In [24]: X_val = (X_val + X_val.T) / 2

In [25]: env = dict()

In [26]: for (var, var_val) in zip(X.flat, X_val.flat):
    ...:     env[var] = var_val
    ...:
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[26], line 2
      1 for (var, var_val) in zip(X.flat, X_val.flat):
----> 2     env[var] = var_val

RuntimeError: You should not call `__bool__` / `__nonzero__` on `Formula`. If you are trying to make a map with `Variable`, `Expression`, or `Polynomial` as keys (and then access the map in Python), please use pydrake.common.containers.EqualToDict`.
```

cc @jwnimmer-tri

jwnimmer-tri, 2023-10-29:
We should fix `EqualToDict` so that it's accepted as an argument anywhere a `std::map` (aka Python `dict`) is required.  In other words, the problem here is not with the bindings of `Evaluate` but rather with `EqualToDict` itself.

As a work around, don't use `[]` to populate the dict, just give it the full list of pairs in its constructor:

```python
env = dict(zip(X.flat, X_val.flat))
```

hongkai-dai, 2023-10-29:
Thanks Jeremy! Could you show me a starting point on how to fix EqualToDict?

jwnimmer-tri, 2023-10-29:
I don't know.  The terms to search for in the pybind11 docs might be either "type caster" or "implicitly convertible".  Or maybe try to find a similar piece of code that already exists in pydrake somewhere.

closed 2026-02-06 by jwnimmer-tri