RobotLocomotion/drake #20449: Handle Continuous Joints in GcsTrajectoryOptimization

cohnt, 2023-10-30:
If a robot has any continuous revolute joints (or a mobile base), it would be helpful for GCS to naturally handle the 0-2pi wraparound that occurs. We have an extension, ["GGCS"](https://www.roboticsproceedings.org/rss19/p057.html), which can handle this, and I'd like to add it to Drake. I imagine it would require the following components:

- A method to specify which degrees of freedom are continuous revolute.
- A method to check that all sets in the graph respect the convexity radius requirement when projected onto each degree of freedom.
- A method to add the "wraparound" edges, along with their corresponding constraints and objectives.

I'm happy to be the one to implement this, but I wanted to create this issue to track any discussion (and hear from @RussTedrake and @wrangelvid before I start making PRs to their code).

wrangelvid, 2023-10-30:
Great! Looking forward to see the PR's.
- Method to specify revolute dof: A public method on `GcsTrajectoryOptimization` and private methods on `Subgraphs` and `EdgesBetweenSubgraphs` would definitely work. The reasoning behind the private method is that I don't think it makes a lot of sense to have subgraphs where revolute joints can be treated as non continuous joints. 
- Method to add wraparound edges + objective and constraints. Once a user specifies the revolute joint with the method above, this should be handled under the hood.

cohnt, 2023-11-30:
#20509 landing is a big step along the way! The wraparound edge functionality is now available for AddSubgraph In the coming days/weeks, I'll extend this functionality to other parts of GcsTrajectoryOptimization:

- [ ] EdgesBetweenSubgraphs
- [ ] RegionsConnectThroughSubspace

Other miscellaneous todos:

- [ ] Automatically break up sets into smaller pieces if they're larger than the convexity radius.
- [ ] Add bounding box computation as a method of ConvexSet.
- [ ] Helper method to extract the indices of continuous revolute joints from a multibody plant. (Question: should this be a method of MultibodyPlant? Can it just be a function unconnected to the class, kept within the GcsTrajectoryOptimization namespace to respect [this style guide rule](https://drake.mit.edu/styleguide/cppguide.html#Nonmember,_Static_Member,_and_Global_Functions)?)

closed 2024-02-23 by cohnt