RobotLocomotion/drake #20450: Could you please teach me how to integrate Drake with NVIDIA Omniverse Isaac Sim?

miz2miz, 2023-10-30:
I've started studying Drake with the intention of accelerating the automation of industrial sectors using 'virtual space + robots + AI'.

After considering Unity and Unreal Engine as virtual spaces, I've ultimately decided on Omniverse Isaac Sim as the clear choice.

NVIDIA Omniverse Isaac Sim has features, similar to Warp, that utilize the GPU to its fullest for machine learning, and I've determined it to be beneficial due to its capabilities for fast deep reinforcement learning.

However, given Drake's excellence in aspects like 'model notation rules, solution methods for optimization problems, and above all, being open-source', I believe it would be fantastic if they could be integrated.

Currently, I'm using meshcat for various 3D visualizations. Do you know of any good ways to utilize NVIDIA Omniverse Isaac Sim?

While I believe it's possible to utilize it forcefully through Python or C++, it would be interesting if we could integrate with the USD file format used by NVIDIA Omniverse and use Drake as an extension within Omniverse, don't you think?

I'm not in a particular hurry for a response. 

Currently, four members of our company are studying Isaac Sim, and I'm taking the lead in exploring Drake...

miz2miz

sherm1, 2023-10-30:
cc @erwincoumans

erwincoumans, 2023-11-02:
We are exploring different options to integrate Drake (or PyDrake and other physics simulators) into Omniverse. Some discussion is here: https://forums.developer.nvidia.com/t/could-you-please-teach-me-how-to-integrate-drake-with-nvidia-omniverse-isaac-sim/271208
Shall I follow up here when we have a first proof of concept?

miz2miz, 2023-11-03:
Understood. 
y-mizomoto@NvidiaOmniverse=miz2miz@github. 

Let's discuss various things on the Nvidia forum.

jwnimmer-tri, 2023-11-03:
In my opinion, questions about Omniverse should be directed to the NVIDIA forums.  Please continue the discussion there for help with Omniverse.

Drake uses a somewhat modular architecture, so you should be able to run the features you need from Drake and then pass those calculations along to where it needs to go.

Note that a feature request for Drake to load USD files is already tracked at https://github.com/RobotLocomotion/drake/issues/17833.

As stated in our [issue template](https://github.com/RobotLocomotion/drake/issues/new/choose), **questions** about how to **use** Drake should be posted to Stack Overflow, not to our issue database.  Our issues database is used for tasks that require action from maintainers -- either bugs, or feature requests.  Since this seems to neither be a bug nor a feature request, I'm going to mark it closed.

It's still perfectly fine to continue discussions or share information here for anyone who is interested.  Closing the issue doesn't prevent more comments, it's just a signal that the Drake maintainers don't need to take any action here to fix or improve the code.

closed 2023-11-03 by jwnimmer-tri

RussTedrake, 2024-04-02:
FWIW: https://www.linkedin.com/posts/toyota-research-institute_model-based-design-and-verification-for-robotics-activity-7175893227848261632-4J71