RobotLocomotion/drake #20451: Return the probability for the sample in Distribution class

hongkai-dai, 2023-10-30:
In Drake `Distribution` class, currently we support `Sample()` function, https://github.com/RobotLocomotion/drake/blob/40e116d44929301d261f15f4d79c0d29b1e8293f/common/schema/stochastic.h#L203-L213

I think it would be very useful to also return the probability of that sample. It would also be very useful to return that probability as an autodiff scalar, since often cases we will want to differentiate the log probability (as is done in reinforcement learning, for example in policy gradient).

cc @ggould-tri @IanTheEngineer @rcory

ggould-tri, 2023-10-30:
"probability of a sample" does not have a clear meaning between the different distribution classes (uniform, gaussian, bernoulli).  If you mean "value of the probability density function at this value" then it's always infinite for the Bernoulli distribution, for instance, while if you mean "change in the cumulative density function across this value" then it's always zero for the uniform distribution.

Can you clarify what value you expect this function would return?

jwnimmer-tri, 2023-10-30:
My inference was he wants [CalcProbabilityDensity](https://drake.mit.edu/doxygen_cxx/namespacedrake.html#adc2cd15bd61a427917d479a38a1a3580) returned along with the sample.

hongkai-dai, 2023-10-30:
Thanks Jeremy!

I want to get the pdf function, as comupted in `CalcProbabilityDensity`. But one difference is that the gradient I need is not ∂p/∂x, as `CalcProbabilityDensity` does. Instead I want ∂p(x)/∂θ, where θ is the parameter of this distribution (for example the mean/variance of the Gaussian).

It would be more natural to return the log probability `log(p(x))` instead of `p(x)` for me.

jwnimmer-tri, 2023-10-30:
I think there's a fair chance that calling `dist.ToExpression()` might serve you better.  From there, you can compute whatever symbolic jacobian you need.