RobotLocomotion/drake #20455: VTK Patches to Upstream

svenevs, 2023-10-31:
There are a handful of patches being vendored right now that we want to upstream:

- [ ] [`[vtk] Remove a spurious dependency on VTK::utf8`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/common_core_noutf8.patch)
- [ ] [`[vtk] Fix warnings in VTK::CommonCore`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/common_core_warnings.patch)
- [ ] [`[vtk] Fix warning in VTK::CommonDataModel`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/common_data_model_warnings.patch)
- [ ] [`[vtk] Fix PNG reader error reporting`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/io_image_png_messages.patch)
- [ ] [`[vtk] Remove stack trace printing in Rendering::OpenGL2`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/rendering_opengl2_nobacktrace.patch)
    - Questionable, I want to ask internally, personally my vote is to remove that from VTK for the reasons you bypassed it in drake, but I don't know its origin or history.
- [ ]  [`[vtk] vtkXOpenGLRenderWindow throw rather than abort`](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/rendering_opengl2_throw_rather_than_abort.patch) (in order):
    - [ ] Patch for drake with exceptions: https://github.com/RobotLocomotion/drake/pull/20433
    - [ ] Open a VTK MR to fix the inconsistent behavior between exiting and `abort()` calls.
    - [ ] Patch drake to use the previous VTK MR, and add a signal handler to drake.

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2023-11-30:
I just double-checked that the list of patch is correct.

The first five are pretty clear and are good to move forward.

The last one (XOpenGL error) is a bit different now since the summary was written.  It's might not be worth trying to change anything upstream -- but if you like, I would be OK funding the change to remove the `abort()` calls and instead only ever using `vtkErrorMacro` and then an early-`return` statement.  Drake doesn't urgently need that though, so if it's difficult we can skip it.  If it there is interest from the VTK side, I'm happy to be a good citizen and fund that improvement.

mwoehlke-kitware, 2023-12-28:
https://github.com/RobotLocomotion/drake/pull/20710 is related.

kwryankrattiger, 2024-01-04:
This seems to be fixed in master already: [[vtk] Fix warning in VTK::CommonDataModel](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/vtk_internal/patches/common_data_model_warnings.patch)

Here is a pass at removing the stack trace by making it only print in Debug builds: https://gitlab.kitware.com/vtk/vtk/-/merge_requests/10802

jwnimmer-tri, 2024-01-04:
> I think that completes the list then other than me opening a PR to label the DataModel warning patch as unneeded in newer VTK.

Agreed.  (Specifically -- our patch is moot now not because we upstreamed it, but rather because [this commit](https://gitlab.kitware.com/vtk/vtk/-/commit/2ab127a520ab2c689dc113a35d16b0062ca46547) rewrote the code.)

> The abort issue is going to be slightly more involved to fix properly in VTK, so that could a on a wish list for a later date?

It's OK with me if we skip it.  My only reason to do it would be as a good citizen funding open source, we don't actually need it in Drake.

svenevs, 2024-01-10:
#20746 was already open so I went ahead and finalized it.

> It's OK with me if we skip it. My only reason to do it would be as a good citizen funding open source, we don't actually need it in Drake.

Thank you :slightly_smiling_face:  Lets discuss briefly on Thursday, right now my vote would be to bump this back to On Deck, and we'll find the right time to fix the `abort()` stuff.

jwnimmer-tri, 2024-01-10:
I'm happy to keep "clean up OpenGL aborts" on deck, but once that's the last thing remaining let's open a new, more focused issue just for that -- it's really about "Do novel VTK surgery" more than really "upstream some Drake patch".

svenevs, 2024-01-11:
> let's open a new, more focused issue just for that

=> #20767

Thanks!

closed 2024-01-11 by svenevs