RobotLocomotion/drake #20481: Allow for HSV color initialization in `LightParameter` from YAML

IanTheEngineer, 2023-11-06:
Currently, `LightParameter`'s `color` attribute accepts `Rgba` YAML initialization only. Providing an HSV YAML spelling would enable useful things like sampling from [`Gaussian`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1schema_1_1_gaussian.html) distributions over the HSV space when resolving stochastic color parameters for lighting.

This might take the form of something like the following using a YAML tag:
```
color: !Hsv
  hsv: [0.08333, 1.0, 1.0] #  orange
```
vs the current:
```
color: 
  rgba: [1.0, 0.5, 0.0] #  orange, with optional alpha
```

jwnimmer-tri, 2023-11-06:
Assigning to +@zachfang for triage per https://drake.mit.edu/issues.html.

jwnimmer-tri, 2023-11-06:
(From my perspective, this is low priority because teaching the domain-randomization tools to randomize in HSV and then convert to RGB when writing out to YAML is likely to be pretty easy.)