RobotLocomotion/drake #20486: RenderEngineVtk needs to render shadows

SeanCurtis-TRI, 2023-11-06:
Shadows provide invaluable clues to the spatial relationship of objects. If we hope to create synthetic data for training, shadows increase the fidelity of our images and increase their verisimilitude. VTK generally supports casting shadows but `RenderEngineVtk` doesn't yet implement them.

An example of this can be found in [this VTK example](https://examples.vtk.org/site/Cxx/Rendering/Shadows/).

Essentially, it requires changing the rendering pipeline to include "passes" for computing the shadow depth maps and to apply them.

SeanCurtis-TRI, 2023-12-04:
There are two snags here:

1. It is well documented that the shadows go all wonky if the render window isn't square. See [this issue](https://discourse.vtk.org/t/invalid-shadow-shape-affected-by-the-aspect-ratio-of-the-window/12000/3) or [this issue](https://discourse.vtk.org/t/vtk-incorrect-shadows/5057)
    - I tentatively began an effort to accommodate this by rendering to a square window, but then only reading back the rectangular bit that we care about. Although the "reading back" is only on the CPU side; VTK would still be pulling the entire square image back from the GPU -- which is unfortunate.
2. I've recently discovered that PBR objects don't receive shadows (and their ability to cast shadows is erratic).  See [this issue](https://discourse.vtk.org/t/image-based-lighting-e-g-pbr-vs-shadows/12835).
    - This was revealed by enforcing a square image and simply applying the shadow code. PBR materials don't receive shadows from lights (although they do receive illumination). When we introduced environment maps to VTK we added logic that when an environment map is added (a practical prerequisite for using PBR objects), all objects in the scene to get promoted to PBR materials. This has the net result of making *nothing* receive shadows.

Having to chose between shadows (at all) and glTF objects is *not* a good choice.

SeanCurtis-TRI, 2023-12-13:
It turns out that the shadows in VTK was more problematic than documented in the VTK issue linked above. It turned out that there was indeed a defect in VTK when the object materials were slated to use PBR materials. However, I've created a VTK patch to address it. So, we're making progress.

jwnimmer-tri, 2024-02-15:
Could you please update the status here @SeanCurtis-TRI?

In particular I see, #20944.  Does that PR close this issue, or is there still more work?

SeanCurtis-TRI, 2024-02-15:
Thanks for flagging this. That PR has been updated to automatically close this issue.

closed 2024-02-20 by SeanCurtis-TRI