RobotLocomotion/drake #20496: Kinematic Floating Body and Multibody Movement

WenlongLu, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
By kinematic body, I mean a body
1. whose pose and twist are fully controlly by the user
2. has infinite mass, and can interact with the rest of the dynamic world
It is essentially a special static body with non-zero velocity.

A few examples where it can be used for:
1. Model the environment, a conveyor belt, a moving truck, etc.
2. Efficiently simulate a robot movement with controlled motion noise

**Describe the solution you'd like**
Expand the function of Locking to lock velocity as well, not just treat velocity as zeros.
Then, expose ports from MBP for users to set poses and spatial velocities.

**Describe alternatives you've considered**
Lock + SetPoses after each time step. Velocity for contact solve would be wrong.
Even for this, I need ports exposed to set poses.

WenlongLu, 2023-11-08:
Another alternative would be control the bodies with infinitely high gain. @xuchenhan-tri mentioned, even for this, we need extra feature support from Drake.

@sherm1 @jwnimmer-tri fyi

amcastro-tri, 2023-11-08:
Today we support infinite PD gains for 1-dof joints. So if all you wanted is to move your body in xyz, you'd need to create a body model with three prismatic joints. That of course requires the user to write a sligthly more complex SDF/URDF model, but it'd acchieve what you want. 

How important (or not) is to prescribe rotations? Would it be an ok first step to only care about translations?
FYI, APIs for PD controlled actuators are documented in the "Using PD controlled actuators" section of the [MultibodyPlant docs](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html). They are very simple to setup.

WenlongLu, 2023-11-08:
Thank you for this suggestion! Xuchen mentioned the same thing. It does look theoretically the same as locking the velocities, right? I wonder why you would't suggest three prismatic joints + three revolute joints to prescribe full rigid motion?

amcastro-tri, 2023-11-26:
> I wonder why you would't suggest three prismatic joints + three revolute joints to prescribe full rigid motion?

That's defintiely doable. I tend to always implement the simplest thing that gets the job done. But if you also need rotations, your proposed solution with prsimatic and revloute joints should work.