RobotLocomotion/drake #20516: [py mp] MathematicalProgram.AddLinearConstraint prefers sparse overload?

EricCousineau-TRI, 2023-11-10:
### What happened?

I have a torque controller written as a QP in Python.
The code creates dense NumPy matrices for use with `prog.AddLinearConstraint()`.

When profiling, I find that it tends to use the sparse flavor.
I found this by using `py-spy`, and seeing calls into `scipy.sparse` taking up a nontrivial amount of time.

Not sure why this happens; best guess is that the pybind11 binding overload for sparse matrices has better "affinity" (i.e., can accept arguments with no conversion), though not sure why this would be the case.

For now, I can workaround it by patching Drake to not bind those overloads.

\cc @hongkai-dai @jwnimmer-tri 

### Version

5303029410909be210f04f80e75e3ec0e7890c24

jwnimmer-tri, 2023-11-10:
This might a regression due to #20361.  \CC @AlexandreAmice

That PR had unit tests that directly construct a LinearConstraint that (seem to) prove that the dense vs sparse overloading happens correctly.

However, the unit test for MathematicalProgram only (seem to) prove that sparse overloads are correctly selected -- there were no checks added that `is_dense_A_constructed()` was True when adding dense A matrices.

jwnimmer-tri, 2023-11-10:
@EricCousineau-TRI maybe you could share _exactly_ which type you were passing at the call site (and the types of any other args that might affect the overload selection)?

EricCousineau-TRI, 2023-11-10:
Yeah, they were all `np.array(dtype=float64)` for numerics, and `np.array(dtype=object)` (`Variable`). Here's a snapshot of one of the solves: #20521 (1c55fec31c)

EricCousineau-TRI, 2023-11-10:
More compact version: 2c32aeb988

```shj
$ bazel-bin/tmp/repro --num_x=1
Time / iter: 0.0047 ms
$ bazel-bin/tmp/repro --num_x=1 --disable_scipy
Time / iter: 0.0048 ms
$ bazel-bin/tmp/repro --num_x=2
Time / iter: 0.2794 ms
$ bazel-bin/tmp/repro --num_x=2 --disable_scipy
Time / iter: 0.0210 ms
$ bazel-bin/tmp/repro --num_x=13
Time / iter: 0.2816 ms
$ bazel-bin/tmp/repro --num_x=13 --disable_scipy
Time / iter: 0.0257 ms
```

EricCousineau-TRI, 2023-11-10:
^ Above agrees with what Alex posted in #20519 (should've looked there first :facepalm:)

EricCousineau-TRI, 2023-11-10:
From https://github.com/RobotLocomotion/drake/pull/20519#issuecomment-1806494323, seems like issue is that `scipy_stub` in Drake has different behavior than actual `scipy`, at least the version that I have on my system.

jwnimmer-tri, 2023-11-10:
Sounds like we have a repro in hand now.  I'll offer the next step might be to bisect Drake to figure out which commit caused this.

EricCousineau-TRI, 2023-11-10:
I actually now have a pure `pybind11` repro, so the bisection would be against our fork itself:
77c2e56d82

It could either be the upgrade to pybind11, or it could always have been a problem.

Will check on the pybind11 side.

EricCousineau-TRI, 2023-11-10:
(The issue occurs when an overload seems to involve both dense vs. sparse and an additional argument that is `dtype=object`)

EricCousineau-TRI, 2023-11-10:
Reverted pybind11 to commit pre #20362, still fails: bc27895baa

EricCousineau-TRI, 2023-11-10:
Still occurs when rewinding pybind11 fork back two years (at #16140): 72c2658b89

EricCousineau-TRI, 2023-11-10:
Yeah, I'd posit that this has always been a problem, we just didn't have sufficient test coverage (explicitly checking overload behavior + using actual `scipy`).
But it ultimately seems like a bug with `pybind11` (our fork) itself.

EricCousineau-TRI, 2023-11-10:
Attempted to migrate Drake+Bazel repro to pybind11+CMake, but am currently failing to reproduce the same results locally; odd.
https://github.com/RobotLocomotion/pybind11/pull/68

closed 2023-11-11 by EricCousineau-TRI