RobotLocomotion/drake #20527: Failing to add weld constraint to MultibodyPlant

udayanbb, 2023-11-13:
### What happened?

I attempted to add a weld constraint to a `MultibodyPlant` via a call to `AddWeldConstraint`. The call acted as a no-op, and attempting to check the constraint's activation status with `GetConstraintActiveStatus ` resulted in the following error:
```
Failure at multibody/plant/multibody_plant.cc:452 in GetConstraintActiveStatus(): condition 'constraint_active_status.count(id) > 0' failed.
```
I wonder if there's a missing line in the implementation [here](https://github.com/RobotLocomotion/drake/blob/8679fc268adb5c0f658e60e0058aef4b5b5c13db/multibody/plant/multibody_plant.cc#L3097C33-L3097C59), because we're not reading `weld_constraints_specs_ ` into `constraint_active_status_map`. Thanks!

### Version

Installed latest python bindings via pip install manipulation

RussTedrake, 2023-11-13:
Based on a quick look, I agree with this assessment.  Assigning to @joemasterjohn .

joemasterjohn, 2023-11-13:
Thanks and good catch, @udayanbb. #20530 has the fix for this. The PRs for the constraint active status feature and the weld constraint were in flight at the same time, so this seemed to fall through the cracks when they merged.

I'm curious though, what do you mean when you say:
> The call acted as a no-op,

As I see it, the bug is only that you can't turn on/off weld constraints through the active status API. The internal code that makes use of the weld constraint doesn't even have the filter: compare the code taht [adds ball constraints](https://github.com/RobotLocomotion/drake/blob/edd7acf78e51d0afc2b91b796c2c1de6c7d70993/multibody/plant/sap_driver.cc#L506) to the code that [adds weld constraints](https://github.com/RobotLocomotion/drake/blob/edd7acf78e51d0afc2b91b796c2c1de6c7d70993/multibody/plant/sap_driver.cc#L596). So if you add weld constraints to a disrete plant using SAP, the weld constraints should be there and working, just not able to be de-activated at runtime (we have unit tests that confirm this).

Looking at the code now it seems we are also missing the error that should be thrown if a weld constraint is added to a plant using our default solver, TAMSI. Perhaps you are not using SAP? You can enable SAP with:
```
plant.set_discrete_contact_solver(DiscreteContactSolver::kSap);
```
or if you are using the `MultibodyPlantConfig` functions:
```
MultibodyPlantConfig config{.discrete_contact_solver = "sap"};
AddMultibodyPlant(config, builder);
```

closed 2023-11-13 by joemasterjohn

udayanbb, 2023-11-14:
Thanks for the quick fix! You're right, the apparent no-op was because I was using TAMSI. Switching to SAP fixed it.

PR #20530 (2023-11-13): [multibody] Make weld constraint use active status.
Fixes #20527.

The weld constraint was not wired up to use the constraint active status mechanism, this PR adds the missing feature.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20530)