RobotLocomotion/drake #20559: update README

Vinggui, 2023-11-17:
### What happened?

I recently built and installed Drake from source, and I have been trying some of its new features. As I wanted to test the Meshcat's animation API, I found that Acrobot's `Spatula Slip Control` examples had some interesting c++ demonstration of it.

While following its internal [README](https://github.com/RobotLocomotion/drake/blob/master/examples/acrobot/README.md) to execute the example, I used the following command:
```
bazel run //tools:meldis -- --open-window &
bazel run //examples/acrobot:run_passive
```
The problem in this example is that while using `meldis`, the visualizer will not show the Playback control exposed after `PublishRecording()`.

Later, I was told by @DamrongGuoy (thanks) that I should just run the example without the meldis, and access its `localhost:7000`.

### Version

Nightly

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

Vinggui, 2023-11-20:
I believe that any README that has an example saving an animation (like the spatula one) needs to have some **NOTE** to inform that Meldis will not allow the user to see the animation output.

In this Issue, I basically reported my path while trying Drake, which involved following an `Acrobot` example (using meldis), and then trying to run a spatula with the same meldis already running.

DamrongGuoy, 2023-11-28:
@Vinggui I confirm that this is the sequence of events that happened, correct?

1. Users are interested in a typical example like [examples/acrobot](https://github.com/RobotLocomotion/drake/tree/master/examples/acrobot) with [acrobot/README.md](https://github.com/RobotLocomotion/drake/blob/master/examples/acrobot/README.md) that instructed users to run `meldis` and then run the example itself like these:
```
bazel run //tools:meldis -- --open-window &
```
```
bazel run //examples/acrobot:run_passive
```
[image]

2. Then, users are interested in another example: [examples/hydroelastic/spatula_slip_control](https://github.com/RobotLocomotion/drake/tree/master/examples/hydroelastic/spatula_slip_control) because it has "Playback simulation" feature. However, the `meldis -- --open-window` from the previous step does not show Playback control when this command finishes:
```
bazel run //examples/hydroelastic/spatula_slip_control:spatula_slip_control
```
As shown in the following picture, the `Animations` menu in the MeshCat window from `meldis -- --open-window` is empty.
[image]

3. **Trouble shooting**: when users run `spatula_slip_control`, the console message gives the URL of MeshCat that does support Playback.  It is different from the URL of MeshCat from `meldis --open-window`.  For example, the last line of the log messages below has `http://localhost:7002` while the `meldis --open-window` has `http://localhost:7000/`:
```
drake (master)$ bazel run //examples/hydroelastic/spatula_slip_control:spatula_slip_control
INFO: Analyzed target //examples/hydroelastic/spatula_slip_control:spatula_slip_control (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/hydroelastic/spatula_slip_control:spatula_slip_control up-to-date:
  bazel-bin/examples/hydroelastic/spatula_slip_control/spatula_slip_control
INFO: Elapsed time: 0.428s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Running command line: bazel-bin/examples/hydroelastic/spatula_slip_control/spatula_slip_control
[2023-11-27 15:10:48.768] [console] [info] Meshcat listening for connections at http://localhost:7002
```
As shown in the following two pictures, MeshCat window from `meldis --open-window` (`http://localhost:7000/`) has no Playback, but MeshCat window from `spatula_slip_control` console (`http://localhost:7002`) does have Playback when the simulation ends.
[image]
[image]

I propose a solution in PR #20596.  I'll assign you as a reviewer.

DamrongGuoy, 2023-11-28:
@Vinggui I do not know why Reviewable didn't let me assign you as a reviewer of PR #20596.  I asked @joemasterjohn instead.

Thank you for telling us about your experience.

Vinggui, 2023-11-28:
@DamrongGuoy, I agree with your step by step reproduction of the issue.

Thanks for looking into it.

closed 2023-11-29 by DamrongGuoy

DamrongGuoy, 2023-11-30:
We have fixed the README.  Now [spatula_slip_control/README.md](https://github.com/RobotLocomotion/drake/blob/master/examples/hydroelastic/spatula_slip_control/README.md) looks like this:
[image]

PR #20596 (2023-11-29): [examples] Update spatula_slip_control README about visualization
Fix #20559

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20596)