RobotLocomotion/drake #20565: `CalcBoundingBox` for `geometry::Shape` types

dmcconachie-tri, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
I am creating approximate bounding boxes for rigid bodies made up of arbitrary shapes; in order to achieve this I need to get the bounding boxes of each component shape that makes up the rigid body.

**Describe the solution you'd like**
A virtual function in `drake::geometry::Shape` which is then implemented in each concrete `Shape` type.

**Describe alternatives you've considered**
A `else if` tree of `dynamic_cast`s is functional but poor design.

**Additional context**
Some of the `Shape` types will have additional challenges regarding the implementation of such a function, in particular `HalfSpace`, `Convex`. and `Mesh`. For my particular application I don't need these solved, though they could be solved at the same time.

jwnimmer-tri, 2023-11-20:
>  A virtual function in `drake::geometry::Shape` which is then implemented in each concrete `Shape` type.

Another (probably better) option is a namespace function that returns the answer. One current example of this design is the function
```c++
double drake::geometry::CalcVolume(const Shape& shape);
```

rpoyner-tri, 2023-11-20:
Related: #15121

closed 2025-07-14 by SeanCurtis-TRI