RobotLocomotion/drake #20610: Can you put a python example for IRIS relavant example?

caseypen, 2023-11-30:
**Is your feature request related to a problem? Please describe.**
I tried to compile the old repo of IRIS for "[iris-distro](https://github.com/rdeits/iris-distro)", but I could not make it compile successfully. However, I saw that Drake has integrated quite a few IRIS-relevant functions. Can you please put a simple example of those functions to generate convex sets in a 2D plane? Just like the old iris-distro example.

**Describe the solution you'd like**
A jupyter notebook in the tutorial folder will be helpful.

RussTedrake, 2023-11-30:
I think this is a fine request.  For the more advanced configuration-space versions of IRIS, I already have tutorial draft / feature request here: https://github.com/RobotLocomotion/drake/issues/19514 

For IRIS, we have pretty [nice examples in the unit tests](https://github.com/RobotLocomotion/drake/blob/4433ac86e71b2c201d113dab5f016261b61bd395/geometry/optimization/test/iris_test.cc#L57).  Admittedly it's C++, but the python syntax matches it very closely.

https://github.com/RobotLocomotion/drake/blob/4433ac86e71b2c201d113dab5f016261b61bd395/bindings/pydrake/geometry/test/optimization_test.py#L553

caseypen, 2023-12-03:
Actually, I am trying to use the GCS-relevant algorithm on mobile robots' motion planning and I think your set-up method can be modified and applied well. Did you have any set-up simulations for 2D/2.5D (terrain is considered) planes?

caseypen, 2023-12-06:
Thanks for your suggestion, I tried them recently in the open-source code, and it seems to work. Is there any good way to visualize HPolyhedron with matplotlib? Is there any API to get edges and vertices? Thansk!

caseypen, 2023-12-09:
Hi Russ,

I can use IRIS modules to generate convex spaces given sample points in a 2D map with any geometric shapes. I can also use GCS to generate curves on the generated IRIS-regions.  I can provide an example jupyter-notebook case if needed.

In my case, I want to use GCS to plan a trajectory for a mobile robot in 2D plan. So the dimension of the mobile robot should be considered and I want to constrain the robot inside the generated IRIS regions. I will start this for a robot with a convex shape, so can this be implemented somehow with GCS? Is there an example to show this? Or any suggestions if I want to do this with drake modules?

Thanks again for bringing up this nice open-source library.

Chen

RussTedrake, 2023-12-11:
IrisInConfigurationSpace handles this case directly, but with less guarantees than IRIS.  Have you tried using the more general version of the algorithm?

caseypen, 2023-12-11:
Yes, I also looked into similar algorithms in UAV applications, like creating safety flight corridors (SFC) and solving the OCP with an optimization framework. Those solutions normally search a "guided path" in grid/voxel space and then solve the problem in a created SFC, which is more local, but faster. However, I was impressed by the framework of GCS that solves the planning problem in a more global sense efficiently. In my case, the map is pre-known, but multiple vehicles are also under consideration. GCS may solve the problem better? Or GCS just provide a viable solution for planning in a higher degree of configuration space? I'm still exploring. Thanks for replying!! I appreciate it.

caseypen, 2023-12-11:
I'm sorry that I may not answer your question correctly. What do you mean by "general version" of the algorithm?

astrodec32, 2024-04-11:
> Hi Russ,
>
> I can use IRIS modules to generate convex spaces given sample points in a 2D map with any geometric shapes. I can also use GCS to generate curves on the generated IRIS-regions. I can [...]

Hi Chen, can you please provide the jupyter-notebook you mention here? It would be greatly helpful. Thank you!
Michael

Cramer54, 2024-08-26:
Hi Russ, I want to implement a 2D scenario where multiple robots carry an object and transfer from one place to another place with obstacles in the whole space. Here, I want to compare the trajectory generated between PRM, SPRM, and GCS. Do you have any implementation of GCS related to it ?

Thank you, in advance.

RussTedrake, 2024-08-28:
@Cramer54 -- depending on the exact details, I think that should work very well.  For kinematic motion planning, the best source for ever-expanding GcsTrajOpt examples is [here](https://manipulation.mit.edu/trajectories.html#gcs). Given your short description, the first question is whether you are really planning in 2D Euclidean space or whether you are planning in the configuration space of the robots (e.g. if the robots have joints and moving geometries that need to avoid collisions).