RobotLocomotion/drake #20641: This Obj file is invisible in meshcat illustration, but appears in proximity.

RussTedrake, 2023-12-07:
### What happened?

[blackboard.tar.gz](https://github.com/RobotLocomotion/drake/files/13593836/blackboard.tar.gz)

Using the following code:
```
from pydrake.all import ModelVisualizer
v = ModelVisualizer()
v.AddModels('/Users/russt/Downloads/blackboard/blackboard.sdf')
v.Run(loop_once=True)
```
I see an empty scene in the illustration mode, but see the blackboard in the proximity mode.

After checking the mtl file, I noticed that the obj file defines multiple objects.  If I remove all but one object (e.g. simply delete all lines after 652 `o obj2`), then the first object _does_ appear correctly.

Related to #11949 and a slew of other obj/visualization-related issues.

### What operating system are you using?

macOS 13 (Ventura)

### What installation option are you using?

compiled from source code using Bazel

RussTedrake, 2023-12-08:
Note that a temporary work-around is to separate the obj file into multiple objects. I've been doing it manually since it's not difficult.

RussTedrake, 2023-12-09:
Interestingly, on another obj which I suspected had the same problem, I tried 
```
v = ModelVisualizer()
v.AddModels("/Users/russt/Downloads/bow.obj")
v.Run(loop_once=True)
```
This gave me the more helpful error (admittedly reporting the slightly sad state of affairs):
```
RuntimeError: error: The Wavefront obj data defines 53 unique shapes; only 1 allowed.
```
That error is being produced by obj_to_surface_mesh.  Loading it through and sdf/urdf does not trigger this message.

BTW -- it would be nice if that error recommended a workaround!

jwnimmer-tri, 2024-07-09:
As of Drake v1.30.0, I see this:

```
jwnimmer@call-cps:~/tmp$ env/bin/python3 -m pydrake.visualization.model_visualizer ~/Downloads/blackboard/blackboard/blackboard.sdf
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/jwnimmer/tmp/env/lib/python3.10/site-packages/pydrake/visualization/model_visualizer.py", line 173, in <module>
    _main()
  File "/home/jwnimmer/tmp/env/lib/python3.10/site-packages/pydrake/visualization/model_visualizer.py", line 168, in _main
    visualizer.AddModels(item)
  File "/home/jwnimmer/tmp/env/lib/python3.10/site-packages/pydrake/visualization/_model_visualizer.py", line 267, in AddModels
    self._builder.parser().AddModels(**kwargs)
RuntimeError: The OBJ file contains multiple objects; only OBJs with a single object are supported: File name: '/home/jwnimmer/Downloads/blackboard/blackboard/blackboard.obj'
```

We have https://github.com/RobotLocomotion/drake/issues/17266 as a feature request tracking that.

closed 2024-07-09 by jwnimmer-tri