RobotLocomotion/drake #20653: Add utility functions to get triangle soup from mesh files

SeanCurtis-TRI, 2023-12-11:
**Is your feature request related to a problem? Please describe.**

While Drake offers some proximity queries on meshes (e.g., obj), it isn't an exhaustive set. To support these queries, users would need to be able to get the mesh data (typically vertex and triangle data) to support other, arbitrary queries. Drake currently doesn't have any utility to expose that geometry.

What the user *can* do is get the `Mesh` or `Convex` shape and pass the file path to their own instance of a geometry parser.

**Describe the solution you'd like**
We can create a simple (family of) utility function(s) that would take a mesh file path (of a Drake-supported type) and return a `TriangleSurfaceMesh`. This has the vertices and a set of triangles that share vertices to the extent that the underlying mesh does. While the mesh doesn't contain normal information, the triangle winding is such that per-triangle normals can easily be calculated.

Note: this would only support mesh types (and mesh data) that are supported for proximity queries. Currently that is *only* Wavefront .obj. And, even then, it would omit obj-specified vertex normals, groups, materials, etc. It would simply be the union of the vertices and triangles (with obj polygons triangulated as necessary) found in the obj.

SeanCurtis-TRI, 2023-12-11:
cc @calderpg-tri

xiao-wang007, 2023-12-14:
It would be great to loop through vertices, get vertices coordinates in body frame, get the normals of the surfaces sharing a vertex

jwnimmer-tri, 2024-03-20:
FYI As of #21061, the convex hull's polygon mesh is available as public SceneGraph-related API.  If the user only needed the hull (e.g., if the geometry was already convex, so the hull is the same as a mesh), then this might already be solved.