RobotLocomotion/drake #20662: access to `drake:collision_filter_group` parsed in URDFs

sadraddini, 2023-12-12:
**Is your feature request related to a problem? Please describe.**
`drake:collision_filter_group` are useful in urdfs to filter collisions between the declared group and _previously declared groups_ (in the same urdf). 

However, there is no way to access these groups after when the urdf files are parsed. This poses some problems:
1. If the user wants to declare a collision filter between `groupA` from `urdfA` and `groupB` from `urdfB`, it has to copy/paste all the members into their code. The `add_collision_filter_group` in `ModelDirective` comes close to a solution`:
```
  - add_collision_filter_group:
      name: foo
      members: [modelA::link1, modelA::link2, ..., modelA::linkN]
      ignored_collision_filter_groups: [modelB::groupB]
```
But one has to still write all members of A by hand. We don't have something like the following, it won't work:
```
  - add_collision_filter_group:
      name: foo
      members: [modelA::groupA]
      ignored_collision_filter_groups: [modelB::groupB]
```
it will throw `RuntimeError: error: body with name 'modelA::groupA' not found` because indeed groupA is not a body, but a group of bodies. If you replace `[modelA::groupA]` with `[modelA::link1]`, it will work (so it understands `ignored_collision_filter_groups: [modelB::groupB]` 

**Describe the solution you'd like**
The solution I **love** is to have something like `std::vector<BodyIndex> GetCollisionFilterGroupByName()`.
It would be so useful not only for default collision filtering, but also for declaring other collision filters using various `CollisionChecker` objects, or declare, for instance, CollisionChecker paddings between two groups of bodies. 

For example, in the URDF we would only have:
```
  <drake:collision_filter_group name="foo">
    <drake:member link="link1"/>
    <drake:member link="link 2"/>
    ...
    <drake:member link="linkN"/>
  </drake:collision_filter_group>
```
And then one can grab the bodies in `foo` and `foo2` in the API to do whatever they want to do with it.

**Describe alternatives you've considered**
`AddCollisionFilterGroup` in ModelDirectives would have a field `group_members`
```
  - add_collision_filter_group:
      name: foo
      members: []
      group_members: [modelA::groupA]
      ignored_collision_filter_groups: [modelB::groupB]
```
And this will filter all bodies in `modelA::groupA` with `modelB::groupB`

rpoyner-tri, 2023-12-12:
Thanks for filing the issue. In my mind, there are two maybe independent proposals.

1. Allow `members` or some new field like `group_members` to understand group names in addition to individual link names.
2.  Add features to MultibodyPlant (or SceneGraph?) to remember the names of collision filter groups after parse time.

As it turns out, I had written some notes about (1) a couple of years ago. For posterity's sake, I'll paste them below.

[[paste]]
The existing language requires at least one list of individually enumerated links. It’s a drag to contemplate re-listing individually a bunch of already-grouped links, just to be able to define a rule at a higher layer. To avoid the repetition, consider:
 * Changing the semantics of drake:member to accept either a link name or a group name. If the name provided is ambiguous, (both a group and a link have the name), raise an error.
   * In model directives, this is just new semantics for the members field.

Older, more verbose proposals, maybe still desirable as sugar?
 * Element drake:member_group (scoped group name) – Analogous to the existing [drake:member](https://drake.mit.edu/doxygen_cxx/group__multibody__parsing.html#tag_drake_member), This allows you to insert some already-defined groups into a new group, and then define new relationships using the existing drake:ignored_collision_filter_group language.
   * In model directives, this is just a new field in the existing directive.
 * Element drake:exclude_between_groups (scoped group A, scoped group B) – Given two groups that already exist, just create a new ExcludeBetween() relationship. You could also retroactively create an ExcludeWithin() relationship using this construct and listing the same name twice; it’s not yet clear to me whether that is useful.
   * In model directives, this is a new directive struct.

Of course, spellings and detailed semantics are up for discussion.
[[end-paste]]

Regarding the proposals in the pasted notes, I currently favor the first one -- expand the semantics of `drake:member` and the directives `member` field to allow group names.

As for (2), I don't think it is difficult, but I haven't yet studied the existing API enough to have a proposal.

rpoyner-tri, 2023-12-12:
To be clear, I think it would be acceptable to do either one or both of (1), (2) above. I don't have an opinion yet about their priority.

sadraddini, 2023-12-12:
I think implementing (2) is quite straightforward in `detail_dmd_parser.cc`. We would only add a field to `AddCollisionFilterGroup` called `group_members` or simply `groups`.

EricCousineau-TRI, 2023-12-14:
Assinging @rpoyner-tri per component leads

rpoyner-tri, 2024-02-16:
As of #20925, part (1) identified above is done. I'd like to sketch some thoughts for part (2), and get some feedback.

Let's start with existing collision filters. Because of the idiosyncratic split of philosophy and responsibilities between Mbp and SceneGraph, it is currently nearly impossible to interrogate filtered collision pairs at all. You can ask about a specific pair, and that's it. With perhaps a dozen lines of code, you could regurgitate the filter matrix for geometries, but you would still have to bushwhack to recover a condensed matrix for links/bodies, and you would have no clue of any prior parse-time structure.

One could imagine adding link/body-level filter group information to Mbp, with new/additional/rewritten APIs. I don't have any specific proposals for how these should look, or whether they should modify the existing API. My vague thought is that the thing to preserve is the contents of the named groups and ignore-collision group pairs as they exist after resolution, at the end of parsing. So, any member-group insertions would have already been executed, and the group-insertion graph structure would not be exposed. Note that from Mbp's viewpoint, no filtering changes after Finalize() time. Any SceneGraph level geometry deletions or other shenanigans would not be tracked.

Thoughts? @jwnimmer-tri @SeanCurtis-TRI @sadraddini

jwnimmer-tri, 2024-02-18:
In my reading of the initial request, I did not see any indication at all that Sadra wanted a _live_ regurgitation of the filtering which would incorporate ongoing customization.  My interpretation of the request was _solely_ that we make available the `collision_filter_group` listings as known by the Parser.

If I'm reading correctly, I think that's consistent with your idea "My vague thought is that the thing to preserve is the contents of the named groups and ignore-collision group pairs as they exist after resolution, at the end of parsing."?

I guess the question is where to put that information?

I could imagine we have a `class FilterGroups` that has the full database, with getters for various info.  I don't see any need to have it return `BodyIndex` though -- it can just return the string names for simplicity's sake.  The user can map the names to either indices or bodies or whatever they need.

Do we need to tack that class into MbP?  The plant is already pretty huge.  What if `class Parser` offered a getter for `const FilterGroups&`?  Users who want the parsing information could make a copy of the groups and stash it away where they need it.

Putting it into MbP would imply that any newly-created Parser objects would need to snarf the pre-existing groups and make them available.  In fact, it means the Source Of Truth for the groups would be inside the plant, updated live as we parse (so that multiple parser objects would all contribute to it).

So that's the real question -- is this canonical MbP data, or is it Parser-only data?

> One could imagine adding link/body-level filter group information to Mbp, with new/additional/rewritten APIs.

We have that already -- it's the CollisionChecker filter matrix.  If people want that matrix, they should be using a CollisionChecker.

rpoyner-tri, 2024-02-19:
Since we are talking about a database of string names, it's arguably parser data. The slight awkwardness is that it is currently per-top-level AddModel*() call. If we tack access on to the parser, we will need a way to key into the separate parse events, or a way to merge them.

rpoyner-tri, 2024-03-19:
Having landed 2 PRs to (1) permit construction of groups from member groups, and to (2) report the parsed collision filter group results , I'm declaring victory.

closed 2024-03-19 by rpoyner-tri