RobotLocomotion/drake #20666: Cannot DecomposeAffineExpressions for expressions that should simplify

nepfaff, 2023-12-13:
### What happened?

I'm getting the following error when calling [DecomposeAffineExpressions](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1symbolic.html#ae3c6a31812dc3e509644dbbc95c9778e):
> RuntimeError: While decomposing an expression, we detected a non-polynomial expression: ( - 9.8100000000000005 * (m_0(0) * (0.86845623874571709 * (h_{x_0}(0) / m_0(0)) + 0.49576583321528123 * (h_{z_0}(0) / m_0(0)))) + 4.0155438029981001 * (m_0(0) * (I_{yy_0}(0) / m_0(0)))).

The symbolic engine isn't strong enough to realize that the expression can be simplified and is thus polynomial in the inertial parameters.

I got this error after creating symbolic variables for mass, rotational inertia, and weighted center of mass (h). I then used these to compute rotational unit inertia and center of mass to create a symbolic plant. I then attempted to decompose the inverse dynamics, which are affine in mass, rotational inertia, and weighted center of mass.

### Version

1.21.0

### What operating system are you using?

Ubuntu 22.04

nepfaff, 2023-12-13:
I have been hitting on a different part of symbolic, and I'm getting related issues. This seems to be a problem with the expression simplification not working properly.

The [docs](https://drake.mit.edu/doxygen_cxx/classdrake_1_1symbolic_1_1_expression.html) state that `E / E = 1` simplification is implemented. However, it does not seem to be working here.

I will rename the issue accordingly.

hongkai-dai, 2023-12-13:
@jwnimmer-tri is it ok to assign this issue to you? I can look into the issue as well.

jwnimmer-tri, 2023-12-13:
Sure.

We'll need an actual, concrete reproducer to be able to do anything.

From the quoted test above, I don't see any `E / E` pattern.  If there were `x(0) / x(0)` in the output that would probably indicate a bug.  We make no claim that `x(0) / (x(0) * x(1))` simplifies to `1 / x(1)` for example.  Only when the numerator and denominator are exactly structurally equal, do we simplify.

In case it helps, we do have [pydrake.symbolic.to_sympy](https://drake.mit.edu/pydrake/pydrake.symbolic.html#pydrake.symbolic.to_sympy) (and `from_sympy`) in case SymPy's manipulations can help overcome the failure.

nepfaff, 2023-12-13:
I see. Looks like I didn't take these rules literally enough :)
In that case, what about converting the bug to a feature request? 

Thank you for pointing out the sympy option. I will try that one

jwnimmer-tri, 2023-12-13:
Yeah, we can at least try to see what's happening.

Could you provide some sample code that demonstrates the problem?  Then we can look into it.  Probably the affine decomposition can use more specific rules than the more general symbolic arithmetic.

I would have thought that affine decomposition would have already multiplied through across the nested products.

nepfaff, 2023-12-13:
This one seems good as a minimal example:
```python
from pydrake.all import Variable

x = Variable("x")
y = Variable("y")
expression = x * (y / x)
print(expression)
print(expression.Jacobian([x]))
```

The result is
```
(x * (y / x))
[<Expression "((x * ((-1 * y) / pow(x, 2))) + (y / x))">]
```

jwnimmer-tri, 2023-12-13:
For the record, `Expand` doesn't help either.  It just changes the error message:

```
>>> DecomposeAffineExpressions([expression.Expand()], vars=[y])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: While decomposing an expression, we detected a non-polynomial expression: ((x * y) / x).
```

nepfaff, 2023-12-13:
This one is also an annoying consequence:

```python
expression = x * (y / x)
print(expression.EvaluatePartial({x: 0, y: 1}))
```
It gives
```
RuntimeError: Division by zero: 1/0
```

jwnimmer-tri, 2023-12-13:
BTW Whether or not simplification should change the domain was discussed a lot in #8515.  Some users want to retain the domain failure x=0 across simplification.  Some users want to set aside those details and focus on the simplest algebraic form.

jwnimmer-tri, 2026-09-10:
At the moment, we don't have plans to further improve the symbolic engine.  In general, I suggest using the sympy compatibility functions (`pydrake.symbolic.to_sympy`, etc.) instead for a more full-featured symbolic manipulation engine.

closed 2026-09-10 by jwnimmer-tri