RobotLocomotion/drake #20667: Hyperrectangle::MaybeCalcAxisAlignedBoundingBox Doesn't Catch Unbounded Hyperellipsoid

cohnt, 2023-12-13:
### What happened?

The function `Hyperrectangle::MaybeCalcAxisAlignedBoundingBox` should return nullopt if the set it's called on is unbounded. However, when called on an unbounded hyperellipsoid, specifically on the build "linux-focal-clang-bazel-experimental-release", it does not return nullopt. I first encountered this in [this build](https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-experimental-release/10629/consoleFull), and verified it in [this build](https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-experimental-release/10630/consoleFull).  I'm not able to replicate this behavior on my local machine. The following code snippet can be added to any of the tests to replicate the behavior:
```
Hyperellipsoid H(Eigen::MatrixXd::Zero(2, 2), Eigen::VectorXd::Zero(2));
EXPECT_FALSE(Hyperrectangle::MaybeCalcAxisAlignedBoundingBox(H).has_value());
```

cc @sadraddini 

### Version

170a0f46c6a19684162c5ebd73911b20755f399b

EricCousineau-TRI, 2023-12-14:
Assigning @hongkai-dai per component leads

hongkai-dai, 2023-12-14:
Assign this to @sadraddini as the original author of hyperrectangle

sadraddini, 2023-12-14:
I could replicate this on `focal`. Yikes. Investigating

closed 2023-12-14 by EricCousineau-TRI