RobotLocomotion/drake #20673: Expression: Support pickling

nepfaff, 2023-12-13:
**Is your feature request related to a problem? Please describe.**
I'm computing symbolic expressions that stay constant for my system and computing them takes a very long time. I want to compute them once, store them on disk, and then load them whenever I need them.

**Describe the solution you'd like**
Pickle support for symbolic `Expression`s. Adding pickle support for `Expression` was previously mentioned [here](https://stackoverflow.com/questions/58982413/serializing-drake-objects-with-python).

**Describe alternatives you've considered**
Converting the expressions to `sympy` (using `to_sympy`) and pickling these. This solution works. However, I wanted to create this issue anyway, as going through `sympy` for pickle support seems a bit odd, and spending offline time computing equations seems like a decently common use case. The `sympy` option makes this a low priority.

EricCousineau-TRI, 2023-12-14:
Assigning myself nominally per component leads.
\cc @jwnimmer-tri @hongkai-dai

jwnimmer-tri, 2023-12-15:
FYI When we circle back here -- the implementation tactic should be to reuse the `Unapply` logic from `symbolic_py_unapply.h`, but for pickle we'd want to use `(expression_kind, (args...))` instead of `(ctor_func, (args..))`.  Then when unpickling we'd look up the `ctor_func` based on the `expression_kind`.

j4yyousi, 2025-12-02:
Hi, I realized that to pickle `Expression`s we need to pickle `Variable`s, since `Expression`s can contain `Variable`s. Is it a requirement that `Variable`s must have the same id before and after pickling? Or we can store only `Variable`s’ name and type, then when unpickling reconstruct the `Variable` objects with different ids?

j4yyousi, 2025-12-02:
Why do we need to maintain a bijection between the old and new IDs? 
My understanding is that an ID contains the variable’s type in its most significant byte and a unique integer from a global counter in the remaining 7 bytes. So storing the variable’s name and type would be sufficient, since the global counter will assign a new integer to the reconstructed variable.  Also, my assumption here is that program A will pickle variables while program B will load them, where each program has its own global counter. Thus I can’t see why it would be useful to maintain a bijection between different program instances.

j4yyousi, 2025-12-07:
Hi, I'm also curious to know why `Variable` [equality is checked by ID](https://github.com/RobotLocomotion/drake/blob/fbf42ba67d0e54a98a47a39bd5eb1d23fa476c77/common/symbolic/expression/variable.h#L86)? Since in Sympy, two variables are equal if their class attributes are equal (it assumes unique names), there is no ID system. Which allows the following pickle/unpickle to easily work in Sympy:
```python
#store_sympy.py
from sympy import symbols, sqrt, pi, evaluate
import pickle
with evaluate(False):
    x_sym, y_sym = symbols('x y')
    expr_to_pickle = sqrt(x_sym**2 + y_sym**2 + pi)
with open("expr.pkl", "wb") as f:
    pickle.dump(expr_to_pickle, f)
```

```python
#load_sympy.py
import sympy
import pickle

z_sym, w_sym = sympy.symbols('z z')
print(z_sym == w_sym) #output: True

with open("expr.pkl", "rb") as f:
    unpickled_expr = pickle.load(f)
x_sym, y_sym = sympy.symbols('x y')
print(x_sym == y_sym) #output: False
values_dict = {x_sym: 3, y_sym: 4} #works
print(unpickled_expr.evalf(subs=values_dict))
```