RobotLocomotion/drake #20681: Add `solver_id` as an optional argument in solvers::Solve()

hongkai-dai, 2023-12-15:
I propose we change the `solvers::Solve()` API to add `solver_id`
```cc
MathematicalProgramResult (const MathematicalProgram& prog, const std::optional<Eigen::VectorXd>& initial_guess, const std::optional<SolverOptions>& solver_options, const std::optional<SolverId>& solver_id);
```
I often find that I write some functions within which I need to solve an optimization problem, and I want to allow the caller of that function to specify the solver if they want, for example in PR https://reviewable.io/reviews/RobotLocomotion/drake/20664#-. Currently I do this
```cc
void Foo(const std::optional<SolverId>& solver_id) {
  // Build the optimization program
  ....

  if (solver_id.has_value()) {
    // Solve with the specified solver.
    auto solver = solvers::MakeSolver(solver_id.value());
    solver.Solve(prog, std::nullopt, std::nullopt, &result);
  } else {
    // Drake chooses the solver automatically.
    result = solvers::Solve(prog);
  }
}
```
This `if (solver_id.has_value())` pattern repeats so often in my code, that I think it is better to add a function.

@jwnimmer-tri what do you think?

AlexandreAmice, 2023-12-15:
I like this a lot, exactly for the reason you say.

sadraddini, 2023-12-15:
This means we must have the option to specify `solver_id` in ALL methods that involve optimization. This includes Toppra, IK, Iris, ... there are many!

jwnimmer-tri, 2023-12-15:
I'll reply after the break.  Generally, I am dubious because when the user wants to hard-code a solver the argument should be `const SolverInterface solver = nullptr`, not `optional<SolverId>`.  Remember -- there are out-of-tree implementations of `SolverInterface` which are not known to `MakeSolver(const SolverId& id)`.  The id is generally intended as a _output_ of the solve ("which solver did you use?"), not an input.

jwnimmer-tri, 2023-12-15:
In the meantime, if you are proposing sugar for "this pattern happens a lot, and we should clean it up" you should be linking to example code on master where the sugar would end up being used.  Otherwise, we'll end up meandering around in the dark, all arguing in different directions.

hongkai-dai, 2024-01-02:
@jwnimmer-tri here are some examples that will benefit from the design
1. Toppra::SolvePathParameterization function in https://reviewable.io/reviews/RobotLocomotion/drake/20664#-
2.  HyperEllipsoid::MinimumUniformScalingToTouch https://github.com/RobotLocomotion/drake/blob/119c7f4c18afb0444a64fa188d24bbc64cf78924/geometry/optimization/hyperellipsoid.cc#L97-L110
3. HPolyhedron::IsInfeasible() https://github.com/RobotLocomotion/drake/blob/119c7f4c18afb0444a64fa188d24bbc64cf78924/geometry/optimization/hpolyhedron.cc#L49-L60

Oftencases we wrap `Solve()` function inside another function (this is used quite often in geometry/optimization). But the result depends on the actual solver. So it is better to allow specifying the solver.

As you mentioned, instead of using `optional<SolverId>`, it is better to use `const SolverInterface*`, so the API should look like
```cc
MathematicalProgramResult (const MathematicalProgram& prog, const std::optional<Eigen::VectorXd>& initial_guess, const std::optional<SolverOptions>& solver_options, const SolverInterface* solver=nullptr);
```

jwnimmer-tri, 2024-01-02:
> Toppra::SolvePathParameterization function in #20664

I don't understand how sugar would help this case.  Here, we're not doing the simple thing of "either use the `SolverInterface*` argument, or else fall back to `ChooseBestSolver`".  Instead, we have a boutique call to `MakeFirstAvailableSolver` with special logic.

> HyperEllipsoid::MinimumUniformScalingToTouch

Ditto.  This logic also dances around with `MakeFirstAvailableSolver`, so I'm not sure what sugar and logic you're proposing.

> HPolyhedron::IsInfeasible() 

I don't understand -- we're not choosing the solver here, we're just setting some options.

---

In short, I'm not following your proposal at all.  Maybe if you posted sample call site patches it would be more clear.