RobotLocomotion/drake #20683: Parsing include tags in xml

jbruedigam-bdai, 2023-12-15:
### What happened?

When parsing Mujoco xmls, `<include>` tags are apparently not parsed.
Maybe as a solutiuon, could you temporarily build the full xml and then parse this temporary xml?

### Version

1.23.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

Unknown / Other

sherm1, 2023-12-21:
Assigned to Rico for disposition

rpoyner-tri, 2024-01-02:
\cc @RussTedrake . Looking at the Mujoco docs, this element is a bit different than anything else we support. A correct implementation would require a preprocessing step that processes and deletes the `<include>` node, replacing it with (a portion of) the tinyxml parse of the included file.

RussTedrake, 2024-01-07:
That seems reasonable to me.  I agree it would be good to support this.

FWIW -- SDF does have an `<include>`, which we now support, via libsdf.  http://sdformat.org/tutorials?tut=composition .

rpoyner-tri, 2024-01-08:
I was aware of SDFormat include, but the implementations will have essentially nothing in common. I'll start investigating this more closely.

PR #21087 (2024-03-06): Adds support for <include> tags in the mujoco parser.
Resolves #20683.

+@rpoyner-tri for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21087)
commit 90aafe4f (2024-03-06): Adds support for <include> tags in the mujoco parser. (#21087)

Resolves #20683.
closed 2024-03-06 by RussTedrake