RobotLocomotion/drake #20685: Installing `drake-dev` appears to affect `mypy` checks outside of `drake` libraries

bhung-bdai, 2023-12-15:
### What happened?

Installing the `drake-dev` Debian package as outlined in the [install page](https://drake.mit.edu/apt.html) results in `mypy` failing and being affected for type checks both within and outside of `pydrake`. For example: 
(Within Drake)
[image]
(Outside Drake)
[image]
These errors only appeared after installing 'drake-dev'; prior, the type checking was all passing. What part of the `drake-dev` install could be causing this? 

### Version

Drake 1.23.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

Unknown / Other

jwnimmer-tri, 2023-12-15:
Please supply more information here so that the bug is actionable.  Specifically, what are the necessary steps for someone else to be able to reproduce the problem?

jwnimmer-tri, 2024-01-04:
Ping @bhung-bdai for more information, please.

closed 2024-01-31 by bhung-bdai