RobotLocomotion/drake #20696: Enable substituting symbolic variables with autodiff variables

nepfaff, 2023-12-19:
**Is your feature request related to a problem? Please describe.**
I have some symbolic expressions stored on disk that I want to use in a non-convex program with autodiff variables. This requires me to substitute the symbolic variables with the autodiff variables.

Minimal toy example:
```python
from pydrake.all import (
    MathematicalProgram,
    SnoptSolver,
    Variable,
)

# Loaded from disk
x = Variable("x")
cost_expression = x**2

prog = MathematicalProgram()
x_vars = prog.NewContinuousVariables(1, "x")

def cost_func(vars):
    var = vars[0]
    return cost_expression.Substitute({x: var})

prog.AddCost(cost_func, vars=x_vars)

# Use Snopt with autodiff as the real problem won't be convex
solver = SnoptSolver()
result = solver.Solve(prog)
```

I have taken this from [this](https://stackoverflow.com/questions/77687354/substitute-symbolic-variables-with-autodiff-variables?noredirect=1#comment136960705_77687354) StackOverflow post.

**Describe the solution you'd like**
Being able to run the above example by enabling the substitution of symbolic variables with autodiff variables.

**Describe alternatives you've considered**
I haven't found any yet.

**Additional context**
I would be happy to help implement this!

nepfaff, 2023-12-20:
Updated the "Describe alternatives you've considered" section with manual evaluation

nepfaff, 2023-12-20:
@jwnimmer-tri suggestion of using `ExpressionCost` works beautifully, as detailed [here](https://stackoverflow.com/a/77694087/11721691).

Should we close this? The only downside of `ExpressionCost` is that the name "cost" makes its use case less obvious in this scenario, and adding it as synthetic sugar to `Expression::Evaluate` might be helpful in that regard.

hongkai-dai, 2023-12-20:
Sure, we can close this issue. 

BTW, evaluating ExpressionCost and ExpressionConstraint with AutoDiffXd is quite slow. I would strongly encourage to write your own constraint/cost which natively supports AutoDiffXd if the computation speed becomes a concern/

closed 2023-12-20 by hongkai-dai