RobotLocomotion/drake #20704: Coupler constraints should be parsed from official SDFormat 1.11 mimic tags

scpeters, 2023-12-21:
**Is your feature request related to a problem? Please describe.**

Since https://github.com/RobotLocomotion/drake/pull/20503 was merged, coupler constraints are now created when parsing an SDFormat file containing custom `drake:mimic` tags, which are treated in a similar manner to the `//joint/mimic` tags supported in the drake URDF parser (see https://github.com/RobotLocomotion/drake/pull/18728, https://github.com/RobotLocomotion/drake/issues/17663) and in the [official URDF documentation](https://wiki.ros.org/urdf/XML/joint#Elements). Recently, `mimic` tags have been added to SDFormat 1.11 under the `//joint/axis` and `//joint/axis2` elements, so I think it would be nice if drake could parse these elements instead of / in addition to the custom `drake:mimic` tags.
The SDFormat syntax for `mimic` differs slightly from URDF in the following ways:

1. The SDFormat `mimic` elements are specified in a `//joint/axis` or `//joint/axis2` element instead of within `//joint`. This allows support for mimicking individual axes of multi-axis joints.
2. The `multiplier` and `offset` parameters are specified as child elements of an SDFormat `mimic` element instead of XML attributes.
3. The SDFormat `mimic` element defines an additional `reference` parameter that applies to the constraint equation as shown below. When `reference` == 0, the specification is identical to the URDF definition.

~~~
follower_position = multiplier * (leader_position - reference) + offset
~~~

**Describe the solution you'd like**

Drake should create coupler constraints based on `//joint/axis/mimic`, `//joint/axis2/mimic`, and `//joint/drake:mimic` tags.

**Describe alternatives you've considered**

The `//joint/drake:mimic` tag could be considered for deprecation if the syntax for the official tags is satisfactory.

**Additional context**

* http://sdformat.org/spec?ver=1.11&elem=joint#axis_mimic
* http://sdformat.org/spec?ver=1.11&elem=joint#axis2_mimic
* http://sdformat.org/tutorials?tut=mimic_proposal&cat=pose_semantics_docs&
* https://github.com/gazebosim/sdformat/pull/1166
* https://github.com/gazebosim/sdformat/pull/1333

sherm1, 2023-12-21:
Assigning to Rico for disposition

jwnimmer-tri, 2023-12-30:
Thanks @scpeters, that's good info and I suspect we'll add the parsing.

Big picture, do you have suggestions for how Drake can keep tabs on additions/changes to the SDFormat specification?  Maybe release notes or a news file for libsdformat releases?  Or any blog page that announces new SDFormat specification versions?  I didn't find anything after some quick searching.

scpeters, 2024-01-02:
> Big picture, do you have suggestions for how Drake can keep tabs on additions/changes to the SDFormat specification? Maybe release notes or a news file for libsdformat releases? Or any blog page that announces new SDFormat specification versions? I didn't find anything after some quick searching.

@jwnimmer-tri thanks for the feedback. We included info about the mimic and auto-inertial calculation features in the [Gazebo Harmonic release announcement](https://community.gazebosim.org/t/gazebo-harmonic-release/2311), but I can see how it would be easy to miss the libsdformat / SDFormat parts of that since that was focused more on Gazebo. I'm discussing with the other libsdformat / SDFormat maintainers about how to better communicate new features and will get back to you.

In addition to more clearly communicating our released features, are you also interested in staying updated on features under development? I believe we pinged some Drake folks for feedback on the SDFormat feature design documents (at github.com/gazebosim/sdf_tutorials) while they were under development, but that was a bit ad hoc. Perhaps a tracking issue for major new features in each planned version would be helpful?

scpeters, 2024-01-02:
I've opened https://github.com/gazebosim/sdformat/issues/1353 to track improving communication of new SDFormat features.

jwnimmer-tri, 2024-01-02:
> In addition to more clearly communicating our released features, are you also interested in staying updated on features under development?

It's a good question.  If you like, you could drop a note in our slack `drakedevelopers.slack.com`, perhaps in the `#ros` channel(?), when anything substantial is happening.  That might be onerous for you, though.

Possibly a better answer is if a few Drake people subscribe to pull request notifications on https://github.com/gazebosim/sdf_tutorials.  That way, we'll get an auto-ping when a new one is opened.  I'll do that myself now, and see how it works.