RobotLocomotion/drake #20709: YAML Serialization support for std::shared_ptr and std::unique_ptr

Levi-Armstrong, 2023-12-28:
Is this currently supported?

rpoyner-tri, 2023-12-29:
No idea what component to assign for this.

Levi-Armstrong, 2023-12-29:
It looks like std::shared_ptr and std::unique_ptr could be handled very similar to std::optional; would you agree?

Levi-Armstrong, 2023-12-30:
Additions to write_archive.h

```cpp
  // For std::shared_ptr.
  template <typename NVP, typename T>
  void DoVisit(const NVP& nvp, const std::shared_ptr<T>&, int32_t) {
    this->VisitSmartPtr(nvp);
  }

  // For std::unique_ptr.
  template <typename NVP, typename T>
  void DoVisit(const NVP& nvp, const std::unique_ptr<T>&, int32_t) {
    this->VisitSmartPtr(nvp);
  }

  // This is used for std::shared_ptr, std::unique_ptr or similar.
  template <typename NVP>
  void VisitSmartPtr(const NVP& nvp) {
    // Bail out if the smart pointer is null.
    if ((*nvp.value()) == nullptr) {
      // Since we are not going to add ourselves to root_, we should not list
      // our name in the visit_order either.  This undoes the addition of our
      // name that was performed by the Visit() call on the shared_ptr<T> or
      // unique_ptr<T> value.
      this->visit_order_.pop_back();
      return;
    }

    // Visit the unpacked shared pointer as if it weren't wrapped in
    // shared_ptr<> or unique_ptr<>.
    this->Visit(yaml_serialization::MakeNameValue(nvp.name(), nvp.value()->get()));

    // The above call to Visit() for the *unwrapped* value pushed our name onto
    // the visit_order a second time, duplicating work performed by the Visit()
    // for the *wrapped* value.  We'll undo that duplication now.
    this->visit_order_.pop_back();
  }
```

jwnimmer-tri, 2023-12-30:
> Is this currently supported?

Not yet.

> It looks like `std::shared_ptr` and `std::unique_ptr` could be handled very similar to std::optional; would you agree?

For `std::unique_ptr`, yes.  Semantically, it's interchangeable with `std::optional`, differing only in memory layout.

I'm less sure about `std::shared_ptr`.  Consider the following struct definition and serialization writing to yaml:

```c++
struct Foo {
  std::shared_ptr<std::string> a;
  std::shared_ptr<std::string> b;
};
Foo foo;
foo.a = std::make_shared<std::string>("bar");
foo.b = foo.a;
SaveYamlString(foo, "foo");
```

Should the output be option 1:

```yaml
foo:
  a: bar
  b: bar
```

Or option 2:

```yaml
foo:
  a: &id1 bar
  b: *id1
```

Per https://yaml.org/spec/1.2.2/#anchors-and-aliases, YAML is specifically designed to represent a _graph_ (not merely a _tree_ like JSON).  People might expect the aliasing within a data structure to be reflected in the output.

On the other hand, maybe #16264 would argue for writing out the flattened tree, at least by default.

Levi-Armstrong, 2023-12-30:
@jwnimmer-tri Thank you for the feedback. I see your point on the anchors and aliases and agree starting with a flattened tree would be the easiest to start with by default. I have updated my code above which has a working implementation based on the optional code for both `std::shared_ptr` and `std::unique_ptr`. The only thing I have not tested yet is when a derived class is stored as a the base class in the smart pointers.

Levi-Armstrong, 2024-01-02:
@jwnimmer-tri and @rpoyner-tri I have been going through the serialization code and see several reference to boost serialization and interested to know why it was not used. This is for my own knowledge and not intended to be a critique as I have been looking for an alternative yaml base serialization library.

jwnimmer-tri, 2024-01-02:
We borrowed the "archive / visit" terminology from Boost.Serialization, but we have very different goals that than library.  Boost.Serialization is for binary serialization (aka pickling).  Our goal with the YAML I/O functions was not to pickle large data sets, but rather to provide a human-readable configuration and debugging language that inter-operates in both C++ and Python.  That being said, the archive / visit pattern (a kind of compile-time reflection) allows for multiple serialization back-ends besides YAML, so anyone who wanted to implement a compact binary encoding could do so, without changing Drake.

jwnimmer-tri, 2024-01-03:
@Levi-Armstrong Could you provide some example use case(s) where `unique_ptr` would be used?  (Or `shared_ptr`?)

Adding support for smart pointers doesn't seem overly difficult (though the Python side would be more subtle), but on the other hand I'm having trouble thinking of a use case where I would ever choose `unique_ptr<Foo>` instead of `optional<Foo>` in a configuration struct.  Isn't `std::optional` uniformly better?  I can't come up with any test case examples that I'd want to write for this case.

That makes me wonder if adding smart pointers might be a trap.

closed 2024-02-23 by jwnimmer-tri