RobotLocomotion/drake #20712: macOS sonoma debug sefgault on common/ad/test/standard_operations_abs2_test.cc

svenevs, 2023-12-29:
First appeared in [https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sonoma-clang-bazel-nightly-debug/18/consoleFull](https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sonoma-clang-bazel-nightly-debug/18/consoleFull), no recurrences yet.  Could possibly be related to #20706, or related orka3 updates and changes in resource provisioning (namely, it seems likely that the job got scheduled with something else that was more resource intensive and ran concurrrently with it, whereas the previous node availability / scheduling may have prevented this).

```
[2:49:42 AM]  ERROR: /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-debug/src/common/ad/BUILD.bazel:74:20: Compiling common/ad/test/standard_operations_abs2_test.cc failed: (Exit 1): wrapped_clang_pp failed: error executing command (from target //common/ad:standard_operations_test) external/local_config_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g '-std=c++11' 'DEBUG_PREFIX_MAP_PWD=.' ... (remaining 126 arguments skipped)
[2:49:42 AM]  
[2:49:42 AM]  Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[2:49:42 AM]  clang++: error: unable to execute command: Segmentation fault: 11
[2:49:42 AM]  clang++: error: clang frontend command failed due to signal (use -v to see invocation)
[2:49:42 AM]  Apple clang version 15.0.0 (clang-1500.1.0.2.5)
[2:49:42 AM]  Target: arm64-apple-macosx14.2
[2:49:42 AM]  Thread model: posix
[2:49:42 AM]  InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
[2:49:42 AM]  clang++: note: diagnostic msg: 
[2:49:42 AM]  ********************
[2:49:42 AM]  
[2:49:42 AM]  PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
[2:49:42 AM]  Preprocessed source(s) and associated run script(s) are located at:
[2:49:42 AM]  clang++: note: diagnostic msg: /var/folders/p6/ggzw_9ys6dq0tkn2hl1d752m0000gn/T/standard_operations_abs2_test-ade9ce.cpp
[2:49:42 AM]  clang++: note: diagnostic msg: /var/folders/p6/ggzw_9ys6dq0tkn2hl1d752m0000gn/T/standard_operations_abs2_test-ade9ce.sh
[2:49:42 AM]  clang++: note: diagnostic msg: Crash backtrace is located in
[2:49:42 AM]  clang++: note: diagnostic msg: /Users/admin/Library/Logs/DiagnosticReports/clang++_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
[2:49:42 AM]  clang++: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
[2:49:42 AM]  clang++: note: diagnostic msg: 
[2:49:42 AM]  
[2:49:42 AM]  ********************
[2:49:42 AM]  Error in child process '/usr/bin/xcrun'. 1
[2:49:43 AM]  [11,447 / 11,598] Linking tools/install/libdrake/libdrake.so; 23s remote-cache, darwin-sandbox ... (6 actions running)
[2:49:44 AM]  INFO: From Linking tools/install/libdrake/libdrake.so:
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/fmt/10.1.1/lib' ignored
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/fmt/10.1.1/lib' ignored
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/fmt/10.1.1/lib' ignored
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/spdlog/1.12.0/lib' ignored
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/fmt/10.1.1/lib' ignored
[2:49:44 AM]  ld: warning: duplicate -rpath '/opt/homebrew/Cellar/ipopt/3.14.13/lib' ignored
[2:49:44 AM]  ld: warning: ignoring duplicate libraries: '-lc++', '-ldrake_lcm', '-ldrake_marker', '-lfmt', '-lipopt', '-lm', '-lspdlog', '-lz'
```

Planned course of action: continue monitoring.  Likely backend resolution: change schedules, or possibly decrease RAM allowed to each VM from 8.0 GB to say 7.5GB (max on node is 16, giving it slightly less than half could possibly help prevent future crashes?).

CC @jwnimmer-tri @BetsyMcPhail @mwoehlke-kitware

svenevs, 2023-12-30:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sonoma-clang-bazel-nightly-debug/20/ was happy.

jwnimmer-tri, 2024-03-16:
Closing until it happens again.

closed 2024-03-16 by jwnimmer-tri